nanoframework/Home #792: ESP32 SSL broken in 1.6.5-preview.455

Ellerbach, 2021-07-13:
### Details about Problem

**Target:** ESP32_WROOM_32

**Firmware image version:** 1.6.5-preview.455

Worked before? If so, with which nanoFramework image version: the versions before

**Device capabilities output:**

### Description

[image]

that's running the basic SSL sample which should work direclety. 

### Detailed repro steps so we can see the same problem

1. Just use the basic SSL client sample and run it

### Other suggested things

### Expected behaviour

This should not raise an exception and just work.

### Screenshot

### Additional context

Work before migration to new IDF version

Ellerbach, 2021-07-13:
Run tests on .453 => SSL broken as well
Run tests o, .429 => SSL working perfectly
So the last IDF migration broke SSL

josesimoes, 2021-07-14:
Update on this: it's working fine with the WROVER KIT

josesimoes, 2021-07-14:
@Ellerbach can you try rebuilding the  ESP32_WROOM_32 **but** setting the `TARGET_VARIANT` build option to an empty string?

AdrianSoundy, 2021-07-18:
The issue is due to available RAM to allocated 2 x 16717 bytes input / output buffers required per SSL connection
Due to the Ram heap being fragmented over various pools there wasn't enough memory in any of the pools for the 2nd 16717 allocation. The biggest pooks was 16505, so ~200 bytes short.

Will investigate further ram usages.  nanoClr seems to require 55K
In the mean time will reduce available managed heap to 90k

Also renamed target WROOM_32_V3 -> WROOM_V3_BLE to match with readme.md etc

closed 2021-07-18 by AdrianSoundy