nanoframework/Home #795: Using Windows Storage the SD Card does not mount on ESP32 Wroom

Dweaver309, 2021-07-14:
**VS version  2019 

**Target ESP32 Dev C

**Device capabilities output)-->:**
System Information
HAL build info: nanoFramework running @ ESP32
  Target:   ESP32_WROOM_32_BLE
  Platform: ESP32

Firmware build Info:
  Date:        Jul 13 2021
  Type:        MinSizeRel build with IDF v3.3.5
  CLR Version: 1.6.5.467
  Compiler:    GNU ARM GCC v5.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  NF.StorageTest, 1.0.0.0
  nanoFramework.Hardware.Esp32, 1.3.3.0
  nanoFramework.Runtime.Events, 1.9.1.0
  Windows.Storage, 1.4.4.0
  mscorlib, 1.10.5.4
  Windows.Storage.Streams, 1.12.2.0
  nanoFramework.System.Text, 1.1.1.0

Native Assemblies:
  mscorlib v100.5.0.12, checksum 0x132BDB6F
  nanoFramework.Runtime.Native v100.0.8.0, checksum 0x2307A8F3
  nanoFramework.Hardware.Esp32 v100.0.7.2, checksum 0x1B75B894
  nanoFramework.Hardware.Esp32.Rmt v100.0.3.0, checksum 0x9A53BB44
  nanoFramework.Hardware.Esp32.Ble v100.0.0.0, checksum 0x1B75B894
  nanoFramework.Devices.OneWire v100.0.3.4, checksum 0xA5C172BD
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.0.1, checksum 0x5A31313D
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x3112D24C
  System.Math v100.0.5.2, checksum 0xC9E0AB13
  System.Net v100.1.3.4, checksum 0xC74796C2
  Windows.Devices.Adc v100.1.3.3, checksum 0xCA03579A
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.4, checksum 0xB6D0ACC1
  Windows.Devices.Gpio v100.1.2.2, checksum 0xC41539BE
  Windows.Devices.I2c v100.2.0.2, checksum 0x79EDBF71
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  Windows.Devices.Pwm v100.1.3.3, checksum 0xBA2E2251
  Windows.Devices.SerialCommunication v100.1.1.2, checksum 0x34BAF06E
  System.IO.Ports v100.1.1.3, checksum 0x61B8380C
  Windows.Devices.Spi v100.1.4.2, checksum 0x360239F1
  System.Device.Spi v100.1.0.0, checksum 0x48031DC5
  Windows.Devices.Wifi v100.0.6.2, checksum 0xA94A849E
  Windows.Storage v100.0.2.0, checksum 0x954A4192

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffe436c  0x00017c00
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x180000     nanoCLR
      1    0x00190000       1      0x230000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x180000 (1536kB)   nanoCLR
  0x00190000    0x230000 (2240kB)   Deployment

Deployment Map
Empty

### Description
The SD card does not mount. The internal storage is working as expected. I am not sure when this started but I think it was working before the firmware upddate yesterday.

The bug can be reproduced using the software attached
[NF.StorageTest.zip](https://github.com/nanoframework/Home/files/6816223/NF.StorageTest.zip)
[NF.HardwarelibTest.zip](https://github.com/nanoframework/Home/files/6816224/NF.HardwarelibTest.zip)

Dweaver309, 2021-07-14:
The NF.HardwarelibTest.zip file was uploaded by mistake. I don't know if the issue is firmware or Nuget Packages

Dweaver309, 2021-07-15:
@AdrianSoundy  Thanks for reopening!

AdrianSoundy, 2021-07-18:
Initial testing shows its a problem with V3.3.5. 
Nothing was changed in nanoCLr in this area for the V3.3.1 -> v3.3.5 migration
Further investigations required

AdrianSoundy, 2021-07-22:
Tested with SPI SD card and the it mount ok now with all firmware versions.
The issue may have been related to the nanoff always using BLE version, now fixed and BLE version having problems when running Ibeacon test code.

Dweaver309, 2021-07-25:
I just tested with ESP32_WROOM_32-1.6.6-preview.501.zip...OK firmware I am still getting the same error Exception System.Exception - CLR_E_VOLUME_NOT_FOUND (1) How do I download a none blue tooth version to test.?

Thanks for all the help

sasha-uk, 2021-07-25:
Hello

I've been having the same issue with mounting SD card on esp32. If it helps...I could only install the non-BLE esp32 target using the below steps but got the same (CLR_E_VOLUME_NOT_FOUND) exception. This is my first attempt at storage/sd cards so can't claim that it did work for me before.

1) download the package manualy from https://cloudsmith.io/~net-nanoframework/repos/nanoframework-images-dev/packages/detail/raw/ESP32_WROOM_32/latest/
2) unzip
3) nanoff --update --target ESP32_WROOM_32 --serialport COM3 --clrfile "C:\your path here\ESP32_WROOM_32-1.6.6-preview.501\nanoCLR.bin

Thanks

Dweaver309, 2021-07-25:
When I tried to load nanoCLR.bin using the above I got header errors. However I was successful using this:
esptool.py --chip esp32 --port COM5 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 		C:/temp/bootloader.bin 0x10000 C:/temp/nanoCLR.bin 0x8000 C:/temp/partitions_4mb.bin

after downloading the zip file to the temp directory. The version was not the blue tooth version but I still get the same error. Thanks for the help!

Dweaver309, 2021-07-25:
[config.txt](https://github.com/nanoframework/Home/files/6874323/config.txt)
This is the configuration after installing the above firmware.

Dweaver309, 2021-08-07:
I compiled the SD card example (sd_card_exampleDoesNotWork.c) from the esp-idf-v3.3.5 directory.
The example failed to mount the SD card. 

I compiled the SD card example from the examples that are installed when PlatformIO is 
installed (sd_card_example_WORKS.c). The SD card mounted and all the file options worked.

I wrote an interop (CPPInteropDoesNotWork.cpp) it compiled without issue. Like the example
it was biased on it failed to mount the SD card.

I tried to write an interop biased on the PlatformIO SD card example. I was not successful 
writing the interop because many of the header files used are missing from our tool chain.

I hope this can hope someone fix the issue https://github.com/nanoframework/Home/issues/795
[CPPInteropDoesNotWork.txt](https://github.com/nanoframework/Home/files/6949883/CPPInteropDoesNotWork.txt)
[sd_card_example_WORKS.txt](https://github.com/nanoframework/Home/files/6949884/sd_card_example_WORKS.txt)
[sd_card_exampleDoesNotWork.txt](https://github.com/nanoframework/Home/files/6949885/sd_card_exampleDoesNotWork.txt)

Dweaver309, 2021-08-07:
I compiled the SD card example (sd_card_exampleDoesNotWork.c) from the esp-idf-v3.3.5 directory.
The example failed to mount the SD card. 

I compiled the SD card example from the examples that are installed when PlatformIO is 
installed (sd_card_example_WORKS.c). The SD card mounted and all the file options worked.

I wrote an interop (CPPInteropDoesNotWork.cpp) it compiled without issue. Like the example
it was biased on it failed to mount the SD card.

I tried to write an interop biased on the PlatformIO SD card example. I was not successful 
writing the interop because many of the header files used are missing from our tool chain.

I hope this can hope someone fix the issue.
[CPPInteropDoesNotWork.txt](https://github.com/nanoframework/Home/files/6949908/CPPInteropDoesNotWork.txt)
[sd_card exampeWorks.txt](https://github.com/nanoframework/Home/files/6949909/sd_card.exampeWorks.txt)
[sd_card_example_DoesnotWork.txt](https://github.com/nanoframework/Home/files/6949910/sd_card_example_DoesnotWork.txt)

Dweaver309, 2021-11-20:
I used this nanoff prompt to update my firmware today:
 nanoff --update --target ESP32_WROOM_32 --serialport COM3 --preview

I changed the pins on my SD Drive so I made minor changes to the sample attached below.

Thanks to Adrian and everyone that solved the issue my SD drive is mounting!!!
[Program.txt](https://github.com/nanoframework/Home/files/7574976/Program.txt)

AdrianSoundy, 2021-11-21:
@Dweaver309 
Please test on latest firmware ESP2_REV0 or ESP32_PSRAM_REV0 and with System.IO.Filesystem to make sure its working.

valoni, 2021-11-21:
@AdrianSoundy 

i have issues from starts of idf 4 with System.IO.Filesystem 

[image]

via SPI  with this board 

[image]

Via IDF 3.3.5 "have no issues with SPI"

AdrianSoundy, 2021-11-21:
@valoni 
Please share code you are using so i can test.

valoni, 2021-11-22:
code used for test 

cs```

using System;
using System.Diagnostics;
using System.Threading;

using nanoFramework.Hardware.Esp32;
using nanoFramework.Runtime.Native;

using System.Device.Gpio;
using System.Runtime;

using System.Text;
using nanoFramework.Networking;
using System.Net;
using System.Net.Sockets;
using Windows.Devices.WiFi;
using nanoFramework.Json;

using System.IO;
using nanoFramework.System.IO.FileSystem;

namespace nf.Blink
{
    public class Program
    {
        private static GpioController s_GpioController;
        private static string MySsid = "jorninet";
        private static string MyPassword = "valoni1324576890";
        private static int CheckSD = 1;
        private static SDCard sdcards;
        public static void Main()
        {
            WiFiTest();

            DateTime dateTime = GetNetworkTime(2);
            Rtc.SetSystemTime(dateTime);

            for(int i=1;i<=10;i++)
            {
                Debug.WriteLine(" Date Time {" + i.ToString() + "}=" + dateTime.ToString());
                Thread.Sleep(50);
            }

            s_GpioController = new GpioController();

            GpioPin led = s_GpioController.OpenPin(
               PinNumber('O', 21),
               PinMode.Output);
            Debug.WriteLine("-----------------------------");
            Debug.WriteLine("Hello from nanoFramework!");
            Debug.WriteLine("-----------------------------");
            Debug.WriteLine($"{ SystemInfo.TargetName } running on { SystemInfo.Platform }.");
            Debug.WriteLine("-----------------------------");
            int cnt = 0;

            var myDateJson = JsonConvert.SerializeObject(dateTime).ToString();
            Debug.WriteLine(myDateJson);
            Debug.WriteLine("-----------------------------");

            if(CheckSD==1)
            {
                Debug.WriteLine("----------------------------");
                Debug.WriteLine("---SD CAARD TEST START------");
                Debug.WriteLine("-----------------------------");
                Configuration.SetPinFunction(15, DeviceFunction.SPI1_MOSI);
                Configuration.SetPinFunction(2, DeviceFunction.SPI1_MISO);
                Configuration.SetPinFunction(14, DeviceFunction.SPI1_CLOCK);
[... 303 lines cut ...]
        }

        private static void StorageEventManager_RemovableDeviceInserted(object sender, RemovableDeviceEventArgs e)
        {
            throw new NotImplementedException();
        }

        static int PinNumber(char port, byte pin)
        {

            if (port == 'O')
            {
                return pin;
            }
            else
            { 
            if (port < 'A' || port > 'J')
               throw new ArgumentException();
            }

            return ((port - 'A') * 16) + pin;
        }
    }
}
```

josesimoes, 2021-12-11:
This should be fixed now. Please reopen if not.

valoni, 2021-12-11:
are you sure it fix SD SPI on ESP32 "we are not talking about STM32 xx"

valoni, 2021-12-11:
just tested again on ESP32 it have same issues as described above ,it was used via SPI mode SD CARD it restart CLR 
- tested again 
1.ESP32 VROWER KIT
2.DO IT ESP32
3.TTGO T8 

all have same behavior (if SD CARD used via SPI)

AdrianSoundy, 2021-12-13:
There are some differences between how API works on IDF V3.35 and V4.3.1
The native code needs to be updated to use the SD SPI master driver so that SPI bus can be shared between SD cards and other SPI devices, This will probably need a change to manage code api.   Maybe be able to make it also backwards compatible.

valoni, 2022-03-02:
it should close - because it now work with IDF 4.3.2 and IDF 4.4