nanoframework/Home #796: Reference to the CoreLibrary in UnitsNet.nanoFramework.Pressure library not in sync with nanoFramework.Iot.Device.Bmxx80

vgolovanov, 2021-07-16:
### Details about Problem

**nanoFramework area:** ( Nuget packages )

**VS version:** (2019 )

**Device capabilities output:**

### Description
Having troubles with nanoFramework.Iot.Device.Bmxx80 library usage from Nuget. Seems that reference to the CoreLibrary in "UnitsNet.nanoFramework.Pressure" not in sync with Iot.Device library.

### Detailed repro steps so we can see the same problem

1. Add the reference from nuget to the nanoFramework.Iot.Device.Bmxx80 library.

2. Getting error : "Unable to resolve dependencies. 'nanoFramework.CoreLibrary 1.10.5' is not compatible with 'nanoFramework.Iot.Device.Bmxx80 1.0.134 constraint: nanoFramework.CoreLibrary (>= 1.10.5)', 'UnitsNet.nanoFramework.Pressure 4.97.1 constraint: nanoFramework.CoreLibrary (>= 1.10.5-preview.18)'."

...

### Other suggested things

### Expected behaviour

#### Sample Project

### Screenshot

### Additional context

### I've made an effort to fix the bug

josesimoes, 2021-07-16:
This is expected... your project seems to be referencing mscorlib 1.10.5 (stable) but UnitsNet.nanoFramework.Pressure 4.97.1 is depending on 1.10.5-preview.18.

You have to keep the versions in sync and reference mscorlib 1.10.5-preview.18. Along with nanoFramework.Iot.Device.Bmxx80 from a matching version.

vgolovanov, 2021-07-16:
Problem is that UnitsNet.nanoFramework.Pressure 4.97.1 library is referenced and used in nanoFramework.Iot.Device.Bmxx80 library. Because libraries used different versions of mscorlib, you can't install nanoFramework.Iot.Device.Bmxx80 by nuget.

josesimoes, 2021-08-15:
With UnitsNet.nanoFramework.NNNN v1.0.153 these are referencing stable versions of our libs, so this shouldn't be a problem anymore. 

Please reopen if not.

closed 2021-08-15 by josesimoes