nanoframework/Home #797: Enable Wire Protocol to be used by Application

alberk8, 2021-07-19:
**nanoFramework area:** (Visual Studio extension | Other tools | Hardware/target board | Community targets | Community contributions | Documentation )

### Is your feature request related to a problem? Please describe.
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
The current USB to UART as implemented via the Wire Protocol (WP) allows for debug messages to be displayed and uploading of new firmware but does not allow for switching the default UART for application use.  
 
### Describe the solution you'd like
A clear and concise description of what you want to happen.
A solution where it is possible to disable the default WP behaviour and switch to be a UART server (receiving a serial connection to receive commands, read and write data).  Also it must have the ability to revert to default behaviour and this should be done via C# code.

### Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.
I have tried the system.io.port but that only allows for connecting to Serial server.  Meaning that nanoframework devices can only initiate the connection.

### Additional context
Add any other context or screenshots about the feature request here.

### Submit a PR with the implementation

If you know how to add whatever you're suggesting, let us know and have a [Pull Request (PR)](https://help.github.com/articles/about-pull-requests/) ready. 
If you are uncertain if the feature will be accepted, outline the proposal to confirm it is viable before commiting time and effort to it.

Even if the feature is a good idea and viable, it may not be accepted since the ongoing effort in maintaining the feature may outweigh the benefit it delivers.

josesimoes, 2021-07-19:
As it's handled today, the configs are "fixed" and part of the build. Making it "dynamic" will increase the code size for a small usage scenario...

Plus, what you are suggesting: control this configuration from C# code would require adding new options to one of existing libraries or come up with a new one. And then handle all this from the native end. It does seem a lot of effort and work for this... Not sure about the effort/benefit...

josesimoes, 2021-07-19:
Not sure what you're describing as "serial server"... there is no such concept of master/slave in serial port API... care to elaborate?
Maybe we are more productive discussing this on Discord?

alberk8, 2021-07-20:
Will discuss this on Discord. May I know which channel?.    Essentially, What I want to achieve is for nanoframework device to collect data from sensors and then sent this data is sent to via UART when another device connects to it (PC).   At present this can be done via TCP (wifi/ethernet).

closed 2021-10-02 by stale[bot]