nanoframework/Home #798: Hardware Unit Test failed to deploy on ESP32 WROOM 32 CP210x

Ellerbach, 2021-07-29:
### Details about Problem

**nanoFramework area:** (C# code)

**VS version:** (2019 )

**VS extension version:** latest

**Target:** ESP32 WROOM 32 CP210x UART driver

**Device capabilities output:**

### Description

The unit test always failed to deploy on the ESP32 WROOM 32 CP210x

[image]

### Detailed repro steps so we can see the same problem

1. Create a Unit Test project
2. Adjust <IsRealHardware>True</IsRealHardware>
3. Run the test

### Other suggested things

Most likely a timing issue with this specific driver

### Expected behaviour

Should just deploy and run the tests on the device.

#### Sample Project

### Screenshot

### Additional context

### I've made an effort to fix the bug

josesimoes, 2021-08-09:
I've tried to reproduce this on a ESP32 Azure IoT Kit without success.

It has a CP210x USB to serial driver. I'm using driver 10.1.10.103:
[image]

I've run the TestFramework POC without any issues:
```
[nanoTestAdapter]: Getting ready to run tests...
[nanoTestAdapter]: Settings parsed
[nanoTestAdapter]: Test sources enumerated
[nanoTestAdapter]: Test group is 'E:\GitHub\nf-nanoFramework.TestFramework\poc\TestOfTestFramework\bin\Debug\NFUnitTest.dll'
[nanoTestAdapter]: Setting up test runner in *** CONNECTED DEVICE ***
[nanoTestAdapter]: Checking device on port COM4.
[nanoTestAdapter]: Found: 1 devices
[nanoTestAdapter]: Getting things with ESP32_WROOM_32 @ COM11
[nanoTestAdapter]: Device connect result is True. Attempt 0/5
[nanoTestAdapter]: Erase deployment block storage. Attempt 0/5.
[nanoTestAdapter]: Erase result is True.
[nanoTestAdapter]: Device status verified as being in initialized state. Requesting to resume execution. Attempt 0/5.
[nanoTestAdapter]: Device has completed initialization.
[nanoTestAdapter]: Device is initialized and ready!
[nanoTestAdapter]: Computing deployment blob.
[nanoTestAdapter]: Added 4 assemblies to deploy.
[nanoTestAdapter]: Adding mscorlib v1.10.5.4 (31668 bytes) to deployment bundle
[nanoTestAdapter]: Adding nanoFramework.TestFramework v1.0.154.0 (5520 bytes) to deployment bundle
[nanoTestAdapter]: Adding nanoFramework.UnitTestLauncher v0.0.0.0 (1288 bytes) to deployment bundle
[nanoTestAdapter]: Adding NFUnitTest v1.0.0.0 (3960 bytes) to deployment bundle
[nanoTestAdapter]: Deploying 4 assemblies to device... Total size in bytes is 42436.
[nanoTestAdapter]: Incrementally deploying assemblies to the device
[nanoTestAdapter]: Erasing block @ 0x00190000...
[nanoTestAdapter]: Assemblies successfully deployed to the device
[nanoTestAdapter]: Resolving.
```

Worth noting that I've run nanoFramework.TestFramework **v1.0.154.0** POC after the build which updated the referenced debugger library. The changes that it brought are fixing a deployment issue, yes, but supposedly unrelated. 

Can you give it another try? Please update your test app to the latest TestFramework so we are in sync. 😉

skigrinder, 2021-08-09:
> image
> cc @skigrinder <https://github.com/skigrinder>

Hi Jose,
Thanks very much for looking into this issue!
Sorry, I posted about the nanoff v1.22.9 update issue before I saw this 
email.
Downloaded a fresh local copy of the JSON nuget and updated the ESP32 
firmware to 1.6.6-preview.530 using nanoff v1.22.7.
Then tried to update the TestFramework nuget to v1.0.154.0 but keep 
getting this:

Error
This project references NuGet package(s) that are missing on this 
computer. Enable NuGet Package Restore to download them.
For more information, see http://go.microsoft.com/fwlink/?LinkID=322105.
The missing file is 
..\packages\nanoFramework.TestFramework.1.0.150\build\nanoFramework.TestFramework.targets. 
nanoFramework.Json.Test    D:\!Mort\nF Github\Json\2021-08-09 nested 
array 
exceptions\nanoFramework.Json-develop\nanoFramework.Json.Test\nanoFramework.Json.Test.nfproj 
69

Is something still out of sync or am I doing something wrong?
Thanks,
    Jim

josesimoes, 2021-08-09:
> ..\packages\nanoFramework.TestFramework.1.0.150\build\nanoFramework.TestFramework.targets. nanoFramework.Json.Test    D:\!Mort\nF Github\Json\2021-08-09 nested array [...]

You have to manually update the reference and path for the nanoFramework nuget. Just search the error message in the nfproj file. It should be the version number, nothing more.

skigrinder, 2021-08-10:
Doh!
Thanks Jose' !
I was able to update nanoff to v1.22.9 this morning and re-flash my 
ESP32 units.
Your new warning to hit the bootloader button is really handy!!

Looks like there are a few manual things that need to be done to get 
unit tests working when dealing with a local copy of a NuGet package:
     Manually modify the nfproj files - change the version number of the 
TestFramework lib (in this case v1.0.154)
     Manually remove the NFUnitTest reference and re-add it using 
version 1.0.0.0

Running unit tests from Test Explorer is still hit-and-miss (although 
ESP32 detection and connection seems to be greatly improved!)
Occasionally, it will get through the 'deploy assemblies' phase but it 
still almost always fails:

        [nanoTestAdapter]: Getting ready to run tests...
        [nanoTestAdapter]: Settings parsed
        [nanoTestAdapter]: Test sources enumerated
        [nanoTestAdapter]: Test group is 'D:\!Mort\nF
        Github\Json\2021-08-09 nested array
        exceptions\nanoFramework.Json-develop\nanoFramework.Json.Test\bin\Debug\NFUnitTest.dll'
        [nanoTestAdapter]: Setting up test runner in *** CONNECTED
        DEVICE ***
        [nanoTestAdapter]: Checking device on port COM7.
        [nanoTestAdapter]: Found: 0 devices
        [nanoTestAdapter]: Checking device on port COM7.
        [nanoTestAdapter]: Found: 0 devices
        [nanoTestAdapter]: Checking device on port COM7.
        [nanoTestAdapter]: Found: 1 devices
        [nanoTestAdapter]: Getting things with ESP32_WROOM_32 @ COM7
        [nanoTestAdapter]: Device connect result is True. Attempt 0/5
        [nanoTestAdapter]: Erase deployment block storage. Attempt 0/5.
        [nanoTestAdapter]: Erase result is True.
        [nanoTestAdapter]: Device status verified as being in
        initialized state. Requesting to resume execution. Attempt 0/5.
        [nanoTestAdapter]: Device has completed initialization.
        [nanoTestAdapter]: Device is initialized and ready!
        [nanoTestAdapter]: Computing deployment blob.
        [nanoTestAdapter]: Added 7 assemblies to deploy.
        [nanoTestAdapter]: Adding mscorlib v1.10.5.4 (31668 bytes) to
        deployment bundle
        [nanoTestAdapter]: Adding nanoFramework.Json v2.1.0.0 (29976
        bytes) to deployment bundle
        [nanoTestAdapter]: Adding nanoFramework.System.Collections
        v1.2.0.3 (3992 bytes) to deployment bundle
        [nanoTestAdapter]: Adding nanoFramework.System.Text v1.1.1.3
        (5832 bytes) to deployment bundle
        [nanoTestAdapter]: Adding nanoFramework.TestFramework v1.0.154.0
        (5520 bytes) to deployment bundle
        [nanoTestAdapter]: Adding NFUnitTest v1.0.0.0 (19580 bytes) to
        deployment bundle
        [nanoTestAdapter]: Adding Windows.Storage.Streams v1.12.2.3
        (6388 bytes) to deployment bundle
        [nanoTestAdapter]: Deploying 7 assemblies to device... Total
        size in bytes is 102956.
        [nanoTestAdapter]: *** ERROR: failed to get device execution
        state, aborting deployment ***
        [nanoTestAdapter]: Deploying assemblies. Second attempt.
        [nanoTestAdapter]: Deployment failed.
        [nanoTestAdapter]: *** ERROR: failed to get device execution
        state, aborting deployment ***
        ========== Test run finished: 24 Tests (0 Passed, 1 Failed, 0
        Skipped) run in 1.3 min ==========

skigrinder, 2021-08-10:
Also, when assembly deployment is successful, it can't find an entry 
point !?!?!

    [nanoTestAdapter]: Getting ready to run tests...
    [nanoTestAdapter]: Settings parsed
    [nanoTestAdapter]: Test sources enumerated
    [nanoTestAdapter]: Test group is 'D:\!Mort\nF Github\Json\2021-08-09
    nested array
    exceptions\nanoFramework.Json-develop\nanoFramework.Json.Test\bin\Release\NFUnitTest.dll'
    [nanoTestAdapter]: Setting up test runner in *** CONNECTED DEVICE ***
    [nanoTestAdapter]: Checking device on port COM7.
    [nanoTestAdapter]: Found: 0 devices
    [nanoTestAdapter]: Checking device on port COM7.
    [nanoTestAdapter]: Found: 0 devices
    [nanoTestAdapter]: Checking device on port COM7.
    [nanoTestAdapter]: Found: 0 devices
    [nanoTestAdapter]: Checking device on port COM7.
    [nanoTestAdapter]: Found: 1 devices
    [nanoTestAdapter]: Getting things with ESP32_WROOM_32 @ COM7
    [nanoTestAdapter]: Device connect result is True. Attempt 0/5
    [nanoTestAdapter]: Erase deployment block storage. Attempt 0/5.
    [nanoTestAdapter]: Erase result is True.
    [nanoTestAdapter]: Device status verified as being in initialized
    state. Requesting to resume execution. Attempt 0/5.
    [nanoTestAdapter]: Device has completed initialization.
    [nanoTestAdapter]: Device is initialized and ready!
    [nanoTestAdapter]: Computing deployment blob.
    [nanoTestAdapter]: Added 7 assemblies to deploy.
    [nanoTestAdapter]: Adding mscorlib v1.10.5.4 (31668 bytes) to
    deployment bundle
    [nanoTestAdapter]: Adding nanoFramework.Json v2.1.0.0 (17428 bytes)
    to deployment bundle
    [nanoTestAdapter]: Adding nanoFramework.System.Collections v1.2.0.3
    (3992 bytes) to deployment bundle
    [nanoTestAdapter]: Adding nanoFramework.System.Text v1.1.1.3 (5832
    bytes) to deployment bundle
    [nanoTestAdapter]: Adding nanoFramework.TestFramework v1.0.154.0
    (5520 bytes) to deployment bundle
    [nanoTestAdapter]: Adding NFUnitTest v1.0.0.0 (19580 bytes) to
    deployment bundle
    [nanoTestAdapter]: Adding Windows.Storage.Streams v1.12.2.3 (6388
    bytes) to deployment bundle
    [nanoTestAdapter]: Deploying 7 assemblies to device... Total size in
    bytes is 90408.
    [nanoTestAdapter]: Incrementally deploying assemblies to the device
    [nanoTestAdapter]: Erasing block @ 0x00190000...
    [nanoTestAdapter]: Assemblies successfully deployed to the device
    [nanoTestAdapter]: Resolving.
    [nanoTestAdapter]:
    Total: (8136 RAM - 90408 ROM - 38716 METADATA)
    [nanoTestAdapter]: AssemblyRef    =       52 bytes (      13 elements)
    [nanoTestAdapter]: TypeRef        =      608 bytes (     152 elements)
    [nanoTestAdapter]: FieldRef       =       16 bytes (       4 elements)
    [nanoTestAdapter]: MethodRef      =     1076 bytes (     269 elements)
    [nanoTestAdapter]: TypeDef        =     1736 bytes (     217 elements)
    [nanoTestAdapter]: FieldDef       =      492 bytes (     242 elements)
    [nanoTestAdapter]: MethodDef      =     2816 bytes (    1406 elements)
    [nanoTestAdapter]: StaticFields   =      360 bytes (      30 elements)
    [nanoTestAdapter]:
    [nanoTestAdapter]: DebuggingInfo  =     1412 bytes
    [nanoTestAdapter]:
    [nanoTestAdapter]: Attributes      =      280 bytes (      35 elements)
    [nanoTestAdapter]: TypeSpec        =       20 bytes (       5 elements)
    [nanoTestAdapter]: Resources Files =        0 bytes (       0 elements)
    [nanoTestAdapter]: Resources       =        0 bytes (       0 elements)
    [nanoTestAdapter]: Resources Data  =        0 bytes
    [nanoTestAdapter]: Strings         =    18506 bytes
    [nanoTestAdapter]: Signatures      =     5220 bytes
    [nanoTestAdapter]: ByteCode        =    32296 bytes
    [nanoTestAdapter]:
    [nanoTestAdapter]: Ready.
    [nanoTestAdapter]: Cannot find any entrypoint!
    [nanoTestAdapter]: Done.
    [nanoTestAdapter]: Waiting for debug commands...
    [nanoTestAdapter]: Tests finished.
    [nanoTestAdapter]: Parsing test results...
    ========== Test run finished: 24 Tests (0 Passed, 0 Failed, 0
    Skipped) run in 50 sec ==========

networkfusion, 2021-10-13:
Needs rechecking against the latest preview firmware!