nanoframework/Home #800: nanoFramework.System.Text.RegularExpressions requires a non-existent dependency

devbotas, 2021-08-02:
### Details about Problem

**nanoFramework area:** (C# code | **Nuget packages** | Community contributions | Documentation )

**VS version:** (2017 | 2019 )

**VS extension version:**

**Target:**

**Device capabilities output:**

### Description
I think it is a typo in NuGet manifest file (dependency version should be 1.2.0, not 11.2.0), see the picture. 

### Screenshot
[image]

josesimoes, 2021-08-02:
Fixed in https://github.com/nanoframework/System.Text.RegularExpressions/commit/167f20ad6a12d1c4e73280fa4c7f3bc25dce5bb7

closed 2021-08-02 by josesimoes

josesimoes, 2021-08-02:
Thanks for reporting and fix suggestion! 👍🏻