nanoframework/Home #802: Moar stable libraries using preview dependencies

devbotas, 2021-08-02:
### Details about Problem

**nanoFramework area:** (Nuget packages)

### Description
Yes, I looked through them **all**.

Here's a list of stable Nuget packages still referencing some preview libraries:
https://www.nuget.org/packages/nanoFramework.Iot.Device.Max44009/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Mcp3xxx/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Lm75/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Amg88xx/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Max31856/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Mlx90614/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Charlieplex/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Max31865/
https://www.nuget.org/packages/nanoFramework.Iot.Device.AD5328/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Apa102/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Ads1115/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Ccs811/
https://www.nuget.org/packages/nanoFramework.NetWorkHelper/1.3.2
https://www.nuget.org/packages/nanoFramework.Iot.Device.Max44009/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Mcp9808/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Hcsr04/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Yx5300/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Bh1745/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Mcp25xxx/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Ak8963/
https://www.nuget.org/packages/nanoFramework.Iot.Device.RotaryEncoder/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Adxl345/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Mbi5027/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Adxl357/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Mpu9250/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Lsm9Ds1/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Hmc5883l/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Max7219/
https://www.nuget.org/packages/nanoFramework.Iot.Device.At24C128C/
https://www.nuget.org/packages/nanoFramework.Iot.Device.ShiftRegister/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Sn74hc595/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Dhtxxx/
https://www.nuget.org/packages/nanoFramework.Hardware.Drivers.Keypad-PCF8574/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Uln2003/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Mfrc522/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Pn532/
https://www.nuget.org/packages/nanoFramework.Iot.Device.Pn5180/

Also accidentally found some which don't even have a stable version, although they may not be ready for prime time yet. Listing anyway (missed a few of those, though):
https://www.nuget.org/packages/nanoframework.System.Net.WebSockets.Server/1.0.0-preview.26
https://www.nuget.org/packages/nanoframework.System.Net.WebSockets.Client/1.0.0-preview.26
https://www.nuget.org/packages/nanoframework.System.Net.WebSockets/1.0.0-preview.26
https://www.nuget.org/packages/nanoFramework.Azure.Devices.Client/1.0.0-preview.25

networkfusion, 2021-09-10:
The other libs mentioned do not have the required github workflow files for auto update of dependencies yet. I can create the required files, but these are probably not ready for stable yet...

networkfusion, 2021-09-10:
The above PR's should close this issue...

closed 2021-11-01 by networkfusion