nanoframework/Home #803: ESP32 Pico board is not recognized by VS Extension

devbotas, 2021-08-02:
### Details about Problem

**nanoFramework area:** ( Visual Studio extension  )

**VS version: **

**VS extension version: (The only) stable**

**Target: ESP32 Pico**

**Firmware image version: 1.6.5.3 (others are also affected)**

**Device capabilities output:**

### Description

ESP32 Pico kit is not recongnized by VS extension and never appears on the Device list.

### Detailed repro steps so we can see the same problem

1. Take a virgin Pico.
2. Flash 1.6.5.3 using nanoff.
3. Done. VS extension never finds it, also flashing is reported successful.

### Other suggested things

### Expected behaviour

VS Extension should recognize the board.

### Screenshot
[image]

### Additional context

Adrian confirmed the issue. He suspects VS extension, because the board itself does send debug information. See message on Discord:
https://discord.com/channels/478725473862549535/481782754674212867/871609660354998343

### Make an effort to fix the bug

Ellerbach, 2021-08-11:
Have you try to unplug after flash and replug or use the rescan button in the extension?
I have a lilygo and no issue, and it is well recognized.

andreboom, 2021-08-12:
Hi Laurent, I did not work, but later discovered it was because I used the ESP32_LILYGO target. When I checked the chip I saw that it was an ESP32-WROVER-B and then tried the ESP32_WROOM_32 and ESP_WROVER_KIT targets, both work and the board then showed up in the explorer.

Ellerbach, 2021-08-12:
All ESP32 targets will show up as ESP32_WROOM_32.
The Lilygo ones are [those](http://www.lilygo.cn/prod_view.aspx?TypeId=50033&Id=1307&FId=t3:50033:3)

Ellerbach, 2021-08-12:
And closing the issue as it seems solved. Please don't hesitate to reopen if needed.

closed 2021-08-12 by Ellerbach

andreboom, 2021-08-12:
This is the [Lilygo ](http://www.lilygo.cn/prod_view.aspx?TypeId=50044&Id=1127&FId=t3:50044:3) I used, that did not show up.

josesimoes, 2021-08-12:
> All ESP32 targets will show up as ESP32_WROOM_32.

@Ellerbach this is where? In Device Explorer in VS? If that's the case, then something is wrong with some targets, because they are supposed to show with their PackageName. There and when looking at the device capabilities output. Please confirm.

andreboom, 2021-08-12:
Yes, in Visual Studio after i flashed board with LILYGO target, (stable and preview) does not show in Explorer.

devbotas, 2021-08-12:
> 
> 
> And closing the issue as it seems solved. Please don't hesitate to reopen if needed.

How is it solved?..

Ellerbach, 2021-08-12:
> How is it solved?..

with this comment:

> both work and the board then showed up in the explorer

Reopening it.

devbotas, 2021-08-17:
I can't use my 769 Discovery board, too... Due to exact same reasons:
[image]

devbotas, 2021-08-24:
I have installed VS and extension on a fresh PC, still the same result. ESP Pico and STM32F769 boards are not detected by the extension...

maikebing, 2022-07-16:
use ESP32_LILYGO can't found devices , 
`nanoff --platform esp32 --serialport COM3 --update --target ESP32_LILYGO`
but use  ESP32_PSRAM_REV3'  , it can be found .

Ellerbach, 2022-07-16:
use `nanoff  --serialport COM3 --update --target ESP32_LILYGO`
platform and target are not working well together. platform will detect the board automatically but it's not what you need in this case.

maikebing, 2022-07-17:
[image]

maikebing, 2022-07-17:
```
09:39:36.630 [NanoDevices: new device arrival COM3]
09:39:36.630 [NanoDevices: candidate nano device COM3]
09:39:36.632 [NanoDevices: checking device  COM3 @ 921600]
09:39:37.639 [NanoDevices: checking device  COM3 @ 460800]
09:39:38.645 [NanoDevices: checking device  COM3 @ 115200]
09:39:41.635 [NanoDevices: checking device  COM3 *** 2nd attempt ***]
09:39:41.636 [NanoDevices: checking device  COM3 @ 921600]
09:39:43.642 [NanoDevices: checking device  COM3 @ 460800]
09:39:45.648 [NanoDevices: checking device  COM3 @ 115200]
09:39:48.331 [NanoDevices: quitting device COM3]
09:39:48.331 [NanoDevices: Serial device enumeration completed. Found 0 devices]
```

maikebing, 2022-07-17:
.NET nanoFramework Firmware Flasher v2.1.7+5330b2d91b

Ellerbach, 2022-07-17:
Stupid additional question: have you physically power it off and on after flashing?
Also, have you retried to find the device by waiting like a minute once it's plugged?

maikebing, 2022-07-18:
Use this command , 
`nanoff  --serialport COM3 --update --target ESP32_LILYGO`
The answer is, I tried. The device is always not found.

But use this command , 
`nanoff --platform esp32 --serialport COM3 --update `
The device can always be found.

maikebing, 2022-07-18:
The device name and model are  T-WATCH-2020_V3.0   400MAH   CH9102F

josesimoes, 2022-07-18:
@maikebing ESP32_LILYGO is different than TWtach. Why are you insisting on using this image if you've found that another one works better?

Also know that there is an experimental version of an image specific for the TWtach. Please find the correct name on the Community Targets repository.

Finally, let's move this conversation to Discord where we can provide better support. Or use Discussions. This issues has been long closed and its being hijacked for another matter. 🙂

maikebing, 2022-07-18:
Sorry ,  I mistakenly thought that ESP32_LILYGO was equivalent to TWtach