nanoframework/Home #805: STM32F411RE and STM32F401CEU "micro python" board support

Ellerbach, 2021-08-14:
**nanoFramework area:** (Community targets)

### Is your feature request related to a problem?

Most of the micro python boards are using STM32F411RE and STM32F401CEU processors. 

Few examples:
* https://github.com/mchobby/pyboard-driver/tree/master/PYBStick
* https://micropython.org/resources/pybv10-pinout.jpg
* https://micropython.org/resources/pyblitev10ac-pinout.jpg
* https://micropython.org/resources/pyblitev10-pinout.jpg

The nucleo64 base images for those boards is not working. Issue is that they seems to require specific activations to be able to load the code properly.
While the code is uploaded properly and accepted, the nanoBooter or nanoCLR do not manage to run at all. No serial port is neither created.

### Describe the solution you'd like

A simple support for those cheap and popular boards offering the love of C# to those boards family.

### Describe alternatives you've considered

No specific alternative.

### Additional context

[image]

### Submit a PR with the implementation

networkfusion, 2021-08-14:
https://github.com/nanoframework/nf-Community-Targets 😜

networkfusion, 2021-08-14:
Is that with the latest images after update to chibios 21.x. can you try with an image prior to that?

valoni, 2021-08-14:
Nucleo Images ar using "UART from ST-LINK" for communication with VS 
those boards are using D+(PA12)/D- (PA11)

so BrainPAD (8 Mhz Clock) is equivalent to  STM32F401 image (what clock need this)
and Electron (12 Mhz clock) is equivalent to STM32F411 image (need to estimate clock = what clock need this)
and GHI Fez Cerb (12 Mhz clock) is equivalent to STM32F405 image (need to estimate clock for 8Mhz)

Ellerbach, 2021-08-23:
I've been doing the work thanks to @valoni for the initial elements. Also created a PR on how to create a custom STM32 community board. See https://github.com/nanoframework/nanoframework.github.io/pull/80. Feedback welcome.
So PR for this board to come soon.

Ellerbach, 2021-08-24:
Close thru PR https://github.com/nanoframework/nf-Community-Targets/pull/166

closed 2021-08-24 by Ellerbach