nanoframework/Home #809: BitConverter.GetBytes bug - return empty bytes from double without explicit casting in a new line

dgGC5, 2021-08-19:
### Details about Problem

**nanoFramework area:** C# code 

### Description
BitConverter.GetBytes returns empty bytes if I do not cast the double value to int using explicit conversion in a separated line. If I cast it explicitly between the parenthesis of mentioned function the values are wrong.

### Detailed repro steps so we can see the same problem

I have attached 4 lines of code in the Sample project paragraph. This snippet is self-explanatory I hope. 
First GetBytes call is not working in nanoFramework, while in pure C# is working. The second call is working for both.

### Expected behavior

Bytes should be returned correctly no matter if the casting is explicit or not.

#### Sample Project

https://dotnetfiddle.net/NPXAvC

		double myValue = 514;
		var bytes = BitConverter.GetBytes((int)myValue);
		var integer = (int)myValue;
		var bytes2 = BitConverter.GetBytes(integer);

josesimoes, 2021-10-13:
Test code 1

```
		double myValue = 514;
		var bytes = BitConverter.GetBytes((int)myValue);
		Console.WriteLine(BitConverter.ToString(bytes));
		
		var integer = (int)myValue;
		var bytes2 = BitConverter.GetBytes(integer);
		Console.WriteLine(BitConverter.ToString(bytes2));
```

Output from dotnet console:
02-02-00-00
02-02-00-00

Output from nanoFramework:
02-02-00-00
02-02-00-00

Test code 2

```
		double myValue = 514;
		var bytes = BitConverter.GetBytes(myValue);
		Console.WriteLine(BitConverter.ToString(bytes));
		
		var integer = (int)myValue;
		var bytes2 = BitConverter.GetBytes(integer);
		Console.WriteLine(BitConverter.ToString(bytes2));
```

Output from dotnet console:
00-00-00-00-00-10-80-40
02-02-00-00

Output from nanoFramework:
00-00-00-00-00-10-80-40
02-02-00-00

This was tested running on an ST_STM32F769I_DISCOVERY with fw of develop branch as of today.

Please clarify what's wrong because I'm not getting it...

dgGC5, 2021-10-15:
I have checked the reproduction once again - on the latest version - and it seems to be fixed. 
We can close the issue. Thank you.

Best Regards

closed 2021-10-15 by dgGC5