nanoframework/Home #810: Add managed call to initialise ENC26j80 network adapter

AdrianSoundy, 2021-08-19:
**nanoFramework area:** (System.Net )

### Is your feature request related to a problem? Please describe.
Currently the low level driver in native code exists for the ENC26J80 but there is no way to start it up. 

### Describe the solution you'd like
Add some method to System.Net to pass in the SPI connection parameters and to initialise the Network interface in LWIP.
Could be generalized to init any SPI network adapter

Requires changes in System.Net and native code.

The ENC26J80 driver exists in nanoframework and can be enabled with the option NF_NETWORKING_ENC28J60

josesimoes, 2021-08-19:
I would prefer to have this working without the need for any managed call (or use intervention).

My thought process:
1) When building the image we already know that it will include support for ENC28J60 so it can be enabled without intervention
2) It requires the SPI pins to use for this. These have to be part of the build, pretty much like what happens with Ethernet support:.

I understand that some people would like to have this flexible to nth level, but we have to make compromises. I would rather stick with a fixed configuration (SPI bus CS pin) than having to go the extra step of passing a config from C# and start it and etc...

valoni, 2021-08-19:
is there's way to simplefying it by addin as an target_nf_ethernet_enc26j80.h inside targets boards source

for example inside 
https://github.com/nanoframework/nf-interpreter/tree/develop/targets/ChibiOS/ST_STM32F429I_DISCOVERY

and there's to define what pins what to use "and voila" ethernet is included for boards 

and then to include on firmware by "enable enc26j80 ethernet on json"  on available spi  by  "Ethernet_ENC26J80":"ON"

AdrianSoundy, 2021-08-19:
Yes you can always hard code it in to the build using build parameters but that means someone has to build there own firmware or you have a fixed configuration and another firmware variety to download.

Ideally both ways should be implemented.

If i think of ESP32, there are so many different boards around you don't know what resources(pins) will be available on any board.
For  ST_STM32F429I_DISCOVERY it may be different.

I was only thinking of a simple call from managed code to pass in SPI parameters and some native code to initialise network adapter.

We can start with build version first.

valoni, 2021-08-20:
@AdrianSoundy  you are right "about free spi resources" that is what exactly we need to point
but as @josesimoes said network on firmware need to work in same as Ethernet is supported now all what we need is to simplefy procedure to build an SPI ethernet on firmware

- API_SPI_Ethernet : "ON" - that be mean enable and include ethernet module for SPI on firmware(and automatically include network stack)

**variant1**
- hardware side - enc26j80 module wire it on available SPI pins  
- managed side - start and use wired SPI on module (or ones you plan) and intialise it for use,similiar to SPI for SD CARD.

or yet simple 
**variant2**
nf_targets_spi_ethernet.h  -> put default configured pins "that everyone can change it by their desire and create firmware"
so both way are ok (to be included in firmware for any one who want to include)

networkfusion, 2024-12-16:
@AdrianSoundy Is this now possible?

networkfusion, 2025-03-07:
I think this needs to be re-visited, given the need for WIZNET 5500 etc.