nanoframework/Home #816: M2Mqtt library leaks RAM

devbotas, 2021-08-31:
### Details about Problem

**nanoFramework area:** (Nuget packages)

**VS version:** (2019 )

**VS extension version:**

**Target:Tested with STM 769, ESP PICO, ESP Wrover kit**

**Device capabilities output:**

### Description
M2Mqtt library leaks RAM if message are published one after another without delay. Adding ```Thread.Sleep(50)``` between the publish call kind of hides the problem. Here a demonstration of three tests: single message, two message without delays, and two message with a pause between them:

[image]

Tested on ESP32 Wrover kit, ESP32 Pico and STM 769 boards, all behave identically. 

M2Mqtt library versions 4.6.1-5.0.1 leak around 200 bytes per message, while in 5.0.2 preview versions leak jumps to a whopping 500-1000 bytes per message.

### Detailed repro steps so we can see the same problem
See the sample app below.

#### Sample Project

Best used with STM 769, because then you don't have to deal with WiFi connection problems. This sample produces the output show in animated GIF above.

```
using nanoFramework.M2Mqtt;
using nanoFramework.M2Mqtt.Messages;
using System;
using System.Text;
using System.Threading;

namespace NFApp5 {
    public class Program {
        public static void Main() {
            MqttClient client = new MqttClient("172.16.0.136");

            string clientId = Guid.NewGuid().ToString();
            client.Connect(clientId);

            uint freeRam = 0;
            client.Publish("temp/test", Encoding.UTF8.GetBytes("single message"), MqttQoSLevel.AtLeastOnce, false);
            for (int i = 0; i < 5; i++) {
                freeRam = nanoFramework.Runtime.Native.GC.Run(true);
                client.Publish("temp/free-ram", Encoding.UTF8.GetBytes(freeRam.ToString("F0")), MqttQoSLevel.AtLeastOnce, false);
                Thread.Sleep(1000);
            }

            client.Publish("temp/test", Encoding.UTF8.GetBytes("two messages without delays"), MqttQoSLevel.AtLeastOnce, false);
            for (int i = 0; i < 5; i++) {
                freeRam = nanoFramework.Runtime.Native.GC.Run(true);
                client.Publish("temp/free-ram", Encoding.UTF8.GetBytes(freeRam.ToString("F0")), MqttQoSLevel.AtLeastOnce, false);
                client.Publish("temp/free-ram", Encoding.UTF8.GetBytes(freeRam.ToString("F0")), MqttQoSLevel.AtLeastOnce, false);
                Thread.Sleep(1000);
            }

            client.Publish("temp/test", Encoding.UTF8.GetBytes("two messages with delay"), MqttQoSLevel.AtLeastOnce, false);
            for (int i = 0; i < 5; i++) {
                freeRam = nanoFramework.Runtime.Native.GC.Run(true);
                client.Publish("temp/free-ram", Encoding.UTF8.GetBytes(freeRam.ToString("F0")), MqttQoSLevel.AtLeastOnce, false);
                Thread.Sleep(50);
                client.Publish("temp/free-ram", Encoding.UTF8.GetBytes(freeRam.ToString("F0")), MqttQoSLevel.AtLeastOnce, false);
                Thread.Sleep(1000);
            }

            Thread.Sleep(Timeout.Infinite);
        }
    }
}
```

devbotas, 2021-08-31:
Forgot to mention, all FWs are 1.7.0.2.

devbotas, 2021-09-01:
@Ellerbach 

1. No, I am not using TLS.
2. How do I do that? I tried running this sample (at 5.0.1 stable and also 5.0.2 preview) https://github.com/nanoframework/nanoFramework.m2mqtt/tree/develop/TestMqtt/Mosquito.TemperatureGauge
 
but that didn't go well...
[image]

Ellerbach, 2021-09-01:
check out the Azure samples. You can either place the text certificate in raw text either in resources, either upload them into the device directly. See here: https://github.com/nanoframework/nanoFramework.Azure.Devices#certificate

devbotas, 2021-09-01:
@Ellerbach Ok I fianlly made it through. It seems that the mosquitto.org certificate in the sample is outdated. I got the latest one from the test.mosquitto.org, and got the sample working.

Generally, the results are the same. I say generally, because test.mosquitto.org is very slow and I had to increase the pauses. Thread.Sleep(50) is not enough in the last part of the test, I increased it to Thread.Sleep(1000), and it was (mostly) ok.

Again, library 5.0.2 behaves much worse than 5.0.1. I also notice the same 216 bytes leaked every time in the test without delays.

Also, I noticed that ESPs need more time between messages, too. It is difficult to say of how much is enough, because of the neverending WiFi connection problems, but definitely Thread.Sleep(50) was not always enough.

It is difficult to believe that nF cannot handle network traffic of such small magnitude. I suspect it is related to the race conditions somewhere in the message acknowledgement state machines. Maybe some dictionary/queue is not properly flushed?..

Ellerbach, 2021-09-01:
Thanks a lot for the investigation. I will look at that and check. I don't see much places where the allocated code will just not get removed/cleaned. And correct, one of the place is the dictionary/queue. That's the obvious place. 
For the performances, if you're using only 0 level, it should be fine. That said, the sent is done asynchronously by the ESP32 and it's buffering there as well.
In your code as you're using AtLeastOnce, for sure the message elements to be awaited are stored in the queue. And the answer is awaited. So the queue won't get emptying before the message will get received or the maximum time allowed.
Anyway, great sample to start the investigation and understand where the issue is.

Ellerbach, 2021-09-01:
Ran with AtMostOnce (so level 0) to check if there are memory leak at the base and there is none:

[image]

So when there is not an answer awaited, the queue behave correctly. That's a start :-)

Ellerbach, 2021-09-01:
I've been using ExactlyOnce (added code at the end with AtMostOnce just to send after a minute the memory:

[image]

Now when AtLeastOnce here is what I get as well:

[image]
[image]

So correct, there is an issue somewhere in the lib, some memory is not properly freed. Queue will be the place where it can happen.

Ellerbach, 2021-09-01:
Made few more tests, seems that this leak is there for quite some time, probably since years. I used a version before I started to clean, improve and add the support of MQTT5. I assumed it just worked fine which is not the case. And for my usage it's ok and didn't saw this one coming as my devices either go to deep sleep, either only uses level 0 (AtMostOnce).
Thanks again for the details, will keep you posted once I4ll have time to deep dive.

Ellerbach, 2021-09-01:
@devbotas making some progress, went thru more cleaning of the lib and perf improvement as well. I still have few messages which stays in one of the queue but it's 1. much faster 2. less memory leaks
Still need to make sure the queue will always be at 0 and clean. Surprisingly, the issue is only on the latest loop with the 2 messages and the little bit of time in the mean time. That may be a race condition or something equivalent which then create this effect of having elements in one of the queue piling while they should not.

[image]

Thanks again. More investigations in few days.

Ellerbach, 2021-09-02:
> Nice to see some progress!
> 
> Looking through the code, I really dislike locking that is going on there. For example, _internalQueue is usually locked, but sometimes it is not:
> https://github.com/nanoframework/nanoFramework.m2mqtt/blob/0a23c489a448d8abbf2c6fc76e0b8c5281bae8db/M2Mqtt/MqttClient.cs#L2110
> 
> The same goes for _inflightWaitHandle. Sometimes it is free, sometimes is locked by _internalQueue:
> https://github.com/nanoframework/nanoFramework.m2mqtt/blob/0a23c489a448d8abbf2c6fc76e0b8c5281bae8db/M2Mqtt/MqttClient.cs#L1107
> 
> There are also instances, where _internalQueue is itself locked by _inflightQueue:
> https://github.com/nanoframework/nanoFramework.m2mqtt/blob/0a23c489a448d8abbf2c6fc76e0b8c5281bae8db/M2Mqtt/MqttClient.cs#L1681
> 
> And, of course, sometimes locks span over hundreds of lines, defying the whole purpose of locking.
> 
> Also, there is _inflightQueue and then _session.InflightMessages which seem to be very related, storing objects of the same context (although I may be wrong about this).
> 
> So it is really difficult to understand, what is locking what. There's no way this inter-cross-locking can work as intended :)

Yes, those are the elements I've been adjusting. I gained a lot in performance by locking only the necessary line and not a block of 100 lines. The session one is not used, it's only if you want to keep the session. Let me create a branch and push my current work. Won't have time today at all to get into this.

Ellerbach, 2021-09-16:
@devbotas I have been spending more time and fixed the memory leak, improved the all up process and performances.

Now every message is sent just once with a proper timeout and management:

[image]

This will go under PR today, just running few more tests to make sure all is ok.

Note that I've been adding an automatic cleaning at every ping. All queues and lists are cleaned on ping as in all cases, they have passed any timeout. This will address the potential edge cases.

closed 2021-09-16 by josesimoes