nanoframework/Home #819: Automatic updater broke samples

devbotas, 2021-09-06:
### Details about Problem

**nanoFramework area:** (C# code )

**VS version:** ( 2019 )

### Description
Bot auto-updated M2Mqtt dependency for MQTT Azure sample. NuGet version changed from `5.0.1-preview.3` to `5.0.1-preview.4`:
https://github.com/nanoframework/Samples/commit/b4f1230c776745903843eee85dfb26075e15b9c0#diff-c32c4394e9f6a4731c8eb758411ad58215f4e5f37db221b1db3e3d33f6966215

However, there's a breaking change in M2Mqtt going between those two versions, and MQT Azure sample now will not compile:

[image]

When it comes to M2Mqtt, there was also a breaking changes somewhere around at 4.6.1-preview.73. 

### Other suggested things
- Disable autoupdater for samples repo. This is a very sensitive repo when it comes to public perception. _It must absolutely always work at all times_, but currently it became a fire-and-forget collection of pretty abandoned projects. When someone commits a sample, there's a decent chance committer actually tested it. When the sample is updated without a human overwatching the process, then the likeliness of breaking something increases dramatically.
- Be more strict about following SemVer. There must have not been a 5.0.1 preview version like that. It should have been named 6.0.0
- Under no circumstances auto-update (x).0.0 to (x+1).0.0 for any project.

josesimoes, 2021-09-06:
Under discussion on [Discord thread](https://discord.com/channels/478725473862549535/478725474491564044/884406440574476319).

devbotas, 2021-09-06:
Excerpt from Discord discussion conclusion:

"(...) we currently in agreement to a) switch from "prerelease" to "stable" for nuget updates, and b) add a build for all samples, so that we recognise a sample has been broken".

networkfusion, 2021-09-06:
For reference: https://discordapp.com/channels/478725473862549535/884406440574476319/884494370160664636

closed 2021-09-12 by josesimoes

josesimoes, 2021-09-16:
For the record: the Azure Pipeline for the samples repository has been setup and the rules added. Now all PRs and merge commits trigger a build for the affected sample(s).