nanoframework/Home #820: ADC definitions for STM32769 Discovery board are partly incorrect

devbotas, 2021-09-07:
### Details about Problem

**Target:** ST_STM32F769I_DISCOVERY

**Firmware image version:** 1.7.0.2

Worked before? If so, with which nanoFramework image version: No

**Device capabilities output:**

### Description

Explained in the picture. Pins A3 and A5 on the board cannot currently be used as ADC channels.

[image]

### Other suggested things
Mistyped ADC controller can be fixed easily, but you cannot simply insert missing ADC2_IN7 channel where it is supposed to be, because this will shift channels going after it, thus breaking someone's program. Append it at the end, like so?..

```
const NF_PAL_ADC_PORT_PIN_CHANNEL AdcPortPinConfig[] = {
    
    // ADC1
    {1, GPIOA, 6, ADC_CHANNEL_IN6},
    {1, GPIOA, 4, ADC_CHANNEL_IN4},
    {1, GPIOC, 2, ADC_CHANNEL_IN12},
    
    // ADC3
    {3, GPIOF, 10, ADC_CHANNEL_IN8},
    {3, GPIOF, 8, ADC_CHANNEL_IN6},

    // these are the internal sources, available only at ADC1
    {1, NULL, 0, ADC_CHANNEL_SENSOR},
    {1, NULL, 0, ADC_CHANNEL_VREFINT},
    {1, NULL, 0, ADC_CHANNEL_VBAT},

    // This was missed in FW <1.7.x.x, so appending at the end of the list:
    {3, GPIOF, 9, ADC_CHANNEL_IN7},
};
```

josesimoes, 2021-09-07:
OK about ADC3 IN8.
Not OK with PF9, this is not in error, that pin is being used for SPI5.

Thanks for reporting.

closed 2021-09-07 by josesimoes

devbotas, 2021-09-07:
And SPI5 is used for what?.. Maybe worth adding a comment inside this definition of why pin A5 cannot be used, to save someone's time in the future?

josesimoes, 2021-09-07:
> And SPI5 is used for what?.. Maybe worth adding a comment inside this definition of why pin A5 cannot be used, to save someone's time in the future?

Do you realize that I could ask the other way around: "and ADC3 channel 7 is used for what?"

There is no justification on the design and pin assignment for all the GPIO pins on the board, why would this one be different?
There is a nuget available with the manged helpers for the ADC that can be referenced on projects to help with this.

Granted the documentation on the targets can be vastly improved. Some of them are blank. 
The core team simply doesn't have enough time and resources to go through all this. Issue #348 is there since 2018 asking for the community help to fix/improve it. You can see the interest that that has raised... 

PS: Really appreciate what you've been doing trying to improve nanoFramework. Really. Can ask you the favour to take a more... positive... approach and tone? 😃 From this end it keeps sounding that everything is wrong, a total mess and people working on this seemed to have make random or poor decisions over and over... You realize that what you see here it's the result of hundreds of thousands of work hours, effort, sleepless nights, weekends, etc from a handful of people, 95% done for free, for the benefit of everyone?

devbotas, 2021-09-07:
I am not negative, sorry if it seems so, maybe slightly tired, because you are actually correct about "everything is wrong, a total mess" :( Everything I touch has all sorts of problems :( Right now, at this very moment, I'm preparing another PR, because ADC sample is (also) broken... I'm not sure I ever found one that works on the first try :(

And it is very difficult to find information, like how the ADC channels are mapped. It is basically a reverse-engineering, digging through firmware.  I ask about SPI5, because 769DISCO present A5 as an analog-first pin, not even mentioning SPI5 in the board datasheet. So hence was my question about SPI5. It is absolutely not clear, what is SPI5, where it is used and why was it prioritized over analog pin?..