nanoframework/Home #821: SpinWait.SpinUntil(milliSeconds) exception thrown

9Rune5, 2021-09-07:
### Details about Problem

**nanoFramework area:** C# code 

**Target:** ESP32

### Description
Calling  SpinWait.SpinUntil(millisecondsTimeout) will always throw an exception.

### Detailed repro steps so we can see the same problem

1. Start new project

2. Add following line: `SpinWait.SpinUntil(500);`
3. Debug the project and step through (or set breapoint after the SpinWait).

### Expected behaviour
Program should sleep for half a second

### Actual behaviour
Hello from nanoFramework!
    ++++ Exception System.ArgumentNullException - CLR_E_ARGUMENT_NULL (1) ++++
    ++++ Message: 
    ++++ System.Threading.SpinWait::SpinUntil [IP: 0000] ++++
    ++++ testsleep.Program::Main [IP: 0010] ++++
Exception thrown: 'System.ArgumentNullException' in mscorlib.dll
An unhandled exception of type 'System.ArgumentNullException' occurred in mscorlib.dll

 Uncaught exception 

### Additional context
The SpinWait.SpinUntil() overload that takes a TimeSpan as parameter works fine (though I question its purpose as the resolution seem no better than Thread.Sleep()).

josesimoes, 2021-09-07:
@9Rune5 `SpinWait()` will have the same effect as `Thread.Sleep()` for anything above 20ms (which is the default thread quantum).
It's only useful if you need anything below that.

9Rune5, 2021-09-07:
I believe what I am observing is that `SpinWait.SpinUntil()` cannot do anything less than 9ms and thus appear to yield similar results as `Thread.Sleep()`. I don't have an oscilloscope to confirm though. `Thread.SpinWait()` OTOH is fast, but I have no idea how much time passes (and as its description points out, the duration will vary between devices). Regardless, it is a separate issue though. I just wish I knew how to construct a repro (Stopwatch.IsHighResolution is not set, so I suspect it does not have the resolution I'd need for that).