nanoframework/Home #823: NetworkHelper is only available to WiFi clients

networkfusion, 2021-09-10:
### Details about Problem

**nanoFramework area:** 
Nuget packages
Samples

**VS version:** (2017 | 2019 )

**VS extension version:**
N/A

**Target:**
Mainly STM32 clients, but also others that use Ethernet.

**Device capabilities output:**
N/A

### Description
Some Samples use a NetworkHelper class. This is dependent on the nuget `Windows.Devices.WiFi`. The code for this should be independent, and/or split out into a seperate nuget with the WiFi parts being extensions of it. This would allow the code to run on all targets.

### Other suggested things

### Expected behaviour
All samples build for all platforms.

#### Sample Project

### Screenshot

### Additional context

### I've made an effort to fix the bug

networkfusion, 2021-09-10:
An appropri
[NetworkHelper.cs.txt](https://github.com/nanoframework/Home/files/7145690/NetworkHelper.cs.txt)
ate class for Ethernet devices is attached.

devbotas, 2021-09-11:
Indeed, creating universal samples with some networking is now pretty much impossible. It would be nice if targets without WiFi (like STM769) would not complain about Windows.Devices.WiFi being deployed.

Ellerbach, 2021-09-15:
What is:

- The current NetworkHelper nuget is purely working without wifi (breaking change)
- Creation of a nuget NetworkHelper.Wifi that will work for both

Will that make things more clear?

ping @josesimoes

josesimoes, 2021-10-13:
> IMHO, I would deprecate https://github.com/nanoframework/nanoFramework.Networking.Sntp lib (nuget) and make it a generic `nanoFramework.Networking` lib. this would include the network helper and other future functionality...

This goes against the project long time policy of keeping features contained and independent so one can use exactly what's required to it's project.

In particular, with network features above the sockets layer, it gets dependent of wifi for ESP32 which dictates the need for separate releases that are platform dependent.

networkfusion, 2021-10-13:
> This goes against the project long time policy of keeping features contained and independent so one can use exactly what's required to it's project.
>
> In particular, with network features above the [...]

In that case, are we in need of a `nanoFramework.Networking.Ethernet` lib with a `.NetworkHelper` class? It is worth getting to the bottom of this sooner rather than later since it affects multiple targets (and samples). Preferably it would be advantageous for the appropriate functions to work against both Ethernet and Wifi clients (without a seperate nuget), since that will require seperate projects (nuget independent), but welcome opinions there...

josesimoes, 2022-01-18:
Implemented with nanoframework/System.Net#216.

closed 2022-01-18 by josesimoes