nanoframework/Home #824: Visual Studio can freeze on program deployment if device watchdog occurs.

networkfusion, 2021-09-10:
### Details about Problem

**nanoFramework area:** 
 Visual Studio extension 

**VS version:**
16.10.x

**VS extension version:**
Latest (as of 2021-09-10)

**Target:**
Occured on OrgPal3

**Firmware image version:**
Latest (as of 2021-09-10)

**Device capabilities output:**
N/A

### Description
When deploying a program, the deployment starts and continues:
[image]
However, if the devices watchdog causes a reboot (which is likely when not using a fresh image), Visual studio will continue to show the above message, but will not be usable (showing `Visual Studio is busy`) until the computer is rebooted.

### Other suggested things

### Expected behaviour

Deploying the firmware should have a timeout, so VS can recover and become usable again (without having to restart the computer).

### Screenshot

### Additional context

### Make an effort to fix the bug

networkfusion, 2021-09-22:
Of note, Erasing the flash of a device using the extension does not reboot it (using an orgpal3). Manually rebooting the device after running the command leads to less issues.

josesimoes, 2021-09-22:
> Of note, Erasing the flash of a device using the extension does not reboot it (using an orgpal3). Manually rebooting the device after running the command leads to less issues.

Thanks for extra feedback. BTW, Erase and Deploy are not supposed to reboot the device. They "only" restart the CLR.

networkfusion, 2021-09-23:
Ah, in that case, the message "Device Reboot confirmed" might be a little misleading, and a physical reboot seems to be required in order for the program to be removed from RAM (e.g. the display does not clear until after)

[image]

josesimoes, 2021-09-23:
There are 4 types of reboots:
- to the platform bootloader (if exists)
- to nanoBooter (if exists)
- reboot CLR only
- hard reboot

That message it's a generic one, fits all. Considering the context and being informational it seemed a bit over detailed to have to explain what kind of reboot is being performed... that can be changed of course. 😅 

As for hard reboot after deployment or debug session start, we been over this several times: it's not possible without loosing connection to the device (on some platforms) and the recovery process would be overly complicated to get in sync again.

Regarding the display: that's a peripheral, out of scope of the CLR. It should be up to the application to reset whatever there is to reset/set on app start. 🤓