nanoframework/Home #825: Handle external blogs that refer to old tools.

networkfusion, 2021-09-10:
### Details about Problem
It is apparent that people are stuggling to get started when using old blogs.
Most are likely due to use of an old tool (nanoFirmwareFlasher).
We should find a way to remove the old package and/or redirect to the new tool.

The current workaroud would be:
dotnet tool uninstall --global nanoFirmwareFlasher
dotnet tool install --global nanoff

**nanoFramework area:** 
nanoff tool.

**VS extension version:**
N/A

**Target:**
All

**Firmware image version:**
All

### Description

networkfusion, 2021-09-10:
This includes our own blog (https://www.nanoframework.net/nanofirmawareflasher-is-here/)

josesimoes, 2021-09-12:
Fixed code snippet on our blog.

josesimoes, 2021-09-12:
I've reached out to Wouter Huysentruit with a dm on Twitter about a month ago, no answer and no editting of the blog post.

josesimoes, 2021-09-12:
All versions of the old nanoFirmwareFlasher nuget packages are labeled as deprecated.

networkfusion, 2021-09-13:
I have added a comment to the blog (which might help)...
[image]