nanoframework/Home #826: Math.Pow requires at least one parameter of type double

networkfusion, 2021-09-13:
### Details about Problem

Math.Pow might need a little work... It requires one of the parameters to be a double... but should be able to handle all parameters to be a float or int (even if converted to a double on output). 

**nanoFramework area:** 
C# code

**VS version
All

**VS extension version:**
All

**Target:**
All

**Device capabilities output:**
N/A

### Description

### Detailed repro steps so we can see the same problem

### Other suggested things

### Expected behaviour

#### Sample Project
See https://github.com/networkfusion/MBN-TinyCLR/commit/d69e52f3d52338eb2f20c7d8e3c5d5214a805c7a

### Screenshot
[image]

### Additional context

### I've made an effort to fix the bug
A work around is to just make one of the parameters into a decimal (e.g. change one of the parameters from `2` to `2.0`)

josesimoes, 2021-09-13:
The NET implementation only has double params: https://docs.microsoft.com/en-us/dotnet/api/system.math.pow?view=net-5.0
That's why we decided to revert our implementation with floats. 

As you've mentioned, it's a simple matter of casting to double. We have to stop making changes on this lib and revert them a few weeks later!

networkfusion, 2021-09-13:
Fair enough. I will label it as "wont fix". But available as a warning to others.

closed 2021-09-13 by josesimoes