nanoframework/Home #828: Samples build is failing

josesimoes, 2021-09-17:
### Details about Problem

**nanoFramework area:** ( Samples )

### Description
- The PR#155 in Samples repo it's failling te build. Code needs to be fixed.

josesimoes, 2021-09-17:
Fixed with nanoframework/Samples#156.

closed 2021-09-17 by josesimoes

ikivanc, 2021-09-17:
Is it possible to double check packages @josesimoes ? I rebased from main and pushed my changes into [my PR](https://github.com/nanoframework/Samples/pull/157)

However I still retrieve build error which is related to `DpsSampleApp`, please can you check
[Failed Build with DPSSampleApp step](https://dev.azure.com/nanoframework/samples/_build/results?buildId=22204&view=logs&j=12f1170f-54f2-53f3-20dd-22fc7dff55f9&t=d1e88c09-dceb-53c1-5579-19c846f3526a) ? How can I address this issue in my PR?

josesimoes, 2021-09-17:
> Is it possible to double check packages @josesimoes ? I rebased from main and pushed my changes into [my PR](https://github.com/nanoframework/Samples/pull/157)
>
> However I still retrieve build error [...]

Just pushed the update with the updated solution. Merging it now.

ikivanc, 2021-09-17:
quick update: I fixed these packages in my PR https://github.com/nanoframework/Samples/pull/157 @josesimoes