nanoframework/Home #831: System.IO.Ports  SerialPort  Close() method call raises Timeout Exception

gligorov, 2021-09-17:
### Details about Problem

Running Samples SerialCommunication code,   first scenario Write - for 5 times  then  call the Close() method on the serial and it throws the timeout exception.

### Description
Close()  call on SerialPort  rasises timeout exception

### Detailed repro steps so we can see the same problem

1.  Run the samples for Serial using System.IO.Ports, modify scenario 1 write to only write few times, then call the Close() method on serial.

### Expected behaviour
There should be no exception raised.

#### Sample Project
run Samples, SerialCommunication as described above.

### Screenshot
See attached.

### Additional context

gligorov, 2021-09-17:
[image]

josesimoes, 2021-09-22:
Reproduced.

josesimoes, 2021-10-13:
Close with nanoframework/nf-interpreter#2084.

closed 2021-10-13 by josesimoes