nanoframework/Home #842: Visual Studio 2019 Extension 2019.7.0.1 not list ESP32 firmware starting from 1.7.1-preview.824

alberk8, 2021-09-30:
### Details about Problem

**Target:** ESP32-WROOM

**Firmware image version:** 1.7.1-preview.824 and above

Worked before? If so, with which nanoFramework image version:
Firmware version 1.7.1-preview.815 and below works.

**Device capabilities output:**

### Description
using nanoff to flush the firmware works without problem but the VS 2019 Device Explorer does not display the ESP32 device and you cannot deploy app.

### Detailed repro steps so we can see the same problem

1.  nanoff --target ESP32_WROOM_32 --serialport COM4 --update --preview

2. Open VS 2019 Device Explorer

...

### Other suggested things

### Expected behaviour
1. The Device Explorer should display the ESP32 device.
2. 
A clear and concise description of what you expected to happen.

### Screenshot

[image]

### Additional context

Add any other context about the problem here.

### Make an effort to fix the bug

Attempt to submit a [Pull Request (PR)](https://help.github.com/articles/about-pull-requests/) that fixes the bug. Include in this PR a test that verifies the fix. If you were not able to fix the bug, a PR that illustrates your partial progress will suffice.

Ellerbach, 2021-09-30:
hum, I don't have this issue. They are all listed, sometimes it takes few attempts like for the M5Stack (I would say about 30 seconds/1 minute or so) but otherwise all ESP32 devices are properly seen with those versions.

alberk8, 2021-10-01:
Looks like latest firmware 1.7.1-preview.853 is working.

josesimoes, 2021-10-01:
@alberk8 good to known. I've been using the latest version without any issues.
Guess it will be OK to close this...

alberk8, 2021-10-01:
Working so I am closing.

closed 2021-10-01 by alberk8