nanoframework/Home #846: Esp32.HighResEventListener wrongly uses EventCategory.Custom leading to it will only randomly be possible to subscribe to timeout event

Henrik98, 2021-10-02:
### Details about Problem
Both CustomEvent and HighResTimer use **EventCategory.Custom** and both are initialized statically. Depending on what order the runtime intializes statics, one of them replaces the other?

**nanoFramework area:** C# code

**VS version:** 2019

**VS extension version:**

**Target:** ESP32

**Device capabilities output:**

### Description
Both CustomEvent and HighResTimer use **EventCategory.Custom** and both are initialized statically. Depending on what order the runtime intializes statics, one of them replaces the other?

nanoFramework.Hardware.Esp32\nanoFramework.Hardware.Esp32\HighResTimer.cs

    public class HighResTimer : IDisposable
    {
        internal Int32 _timerHandle;
        internal bool  _disposedValue = false; // To detect redundant calls

       private static HighResEventListener s_eventListener = new HighResEventListener();

nanoFramework.Hardware.Esp32\nanoFramework.Hardware.Esp32\HighResEventListener.cs

    public HighResEventListener()
    {
       EventSink.AddEventProcessor(EventCategory.Custom, this);
       EventSink.AddEventListener(EventCategory.Custom, this);
    }

nanoFramework.Runtime.Events\nanoFramework.Runtime.Events\CustomEvent.cs

    public static class CustomEvent
    {
    ...

    static CustomEvent()
    {
        CustomEventListener customEventListener = new CustomEventListener();

        EventSink.AddEventProcessor(EventCategory.Custom, customEventListener);
        EventSink.AddEventListener(EventCategory.Custom, customEventListener);
    }

### Detailed repro steps so we can see the same problem

1. Add nuget package:
nanoframework.Hardware.Esp32

2. Create a HighResTimer:
_highResTimer = new HighResTimer();

3. Add a handler for _highResTimer.OnHighResTimerExpired:
      _highResTimer.OnHighResTimerExpired += (a, b) => {
         OnHighResTimerExpired(a, b);
       };

4. Start the timer:
_highResTimer.StartOnePeriodic(1000_000 / _samplingFrequenzy);

5. When timeout occurs an exception occurs if CustomEvent is initialized after HighResEventListener.
++++ Exception System.NullReferenceException - CLR_E_NULL_REFERENCE (3) ++++
    ++++ Message: 
    ++++ nanoFramework.Runtime.Events.CustomEvent+CustomEventListener::OnEvent [IP: 000c] ++++
    ++++ nanoFramework.Runtime.Events.CustomEvent::OnCustomEventFiredCallback [IP: 0012] ++++
    ++++ nanoFramework.Runtime.Events.EventSink::EventDispatchCallback [IP: 0014] ++++

### Other suggested things

### Expected behaviour

#### Sample Project

### Screenshot

### Additional context

### I've made an effort to fix the bug

josesimoes, 2021-10-04:
@Henrik98 you're correct on the fact that the HighResTimer it's using the custom event, thus clashing with the usage of it as a custom event. The fix for this is to add a new event specific for the HighResTimer. 
This is relatively easy to do: envolves adding a new event code for this new event (both in the managed and native code) and reworking the  HighResTimer event listener to use this new one. 
If this is something that your willing to tackle I can guide you though it. Please let me know.

Henrik98, 2021-10-05:
Yes, I can fix it

Henrik98, 2021-10-14:
Hi

Yes that was the solution I was thinking to do :-)

The question was more about the process, how to publish the change for review , requirements on test case etc... 
I haven't worked with pull request before, just gerrit.