nanoframework/Home #850: Device Explorer doesn't find STM32F091 device

msschl, 2021-10-05:
### Details about Problem

**nanoFramework area:** Visual Studio extension

**VS version:** Microsoft Visual Studio Community 2019 Version 16.11.3

**VS extension version:** nanoFramework 2019.6.0.80

**Target:** ST_NUCLEO64_F091RC

**Firmware image version:** v1.7.0.2 and v1.7.1-preview.828

### Description

The Device Explorer doesn't find the device.

```txt
21:13:24.782 [NanoDevices: dropping device in exclusion list COM1]
21:13:24.782 [NanoDevices: new device arrival COM3]
21:13:24.782 [NanoDevices: candidate nano device COM3]
21:13:24.785 [NanoDevices: checking device  COM3 @ 921600]
21:13:25.874 [NanoDevices: checking device  COM3 @ 460800]
21:13:26.970 [NanoDevices: checking device  COM3 @ 115200]
21:13:29.864 [NanoDevices: checking device  COM3 *** 2nd attempt ***]
21:13:29.866 [NanoDevices: checking device  COM3 @ 921600]
21:13:31.952 [NanoDevices: checking device  COM3 @ 460800]
21:13:34.025 [NanoDevices: checking device  COM3 @ 115200]
21:13:37.026 [NanoDevices: quitting device COM3]
21:13:37.026 [NanoDevices: Serial device enumeration completed. Found 0 devices]
```

However, the device is successfully flashed via `$ nanoff --update --target ST_NUCLEO64_F091RC --jtag`:

```txt
$ nanoff --update --target ST_NUCLEO64_F091RC --jtag
nanoFramework Firmware Flasher v1.25.1+62cc30c881
Copyright (C) 2019 .NET Foundation and nanoFramework project contributors

** There is a new version available, update is recommended **

Using cached firmware package
Extracting ST_NUCLEO64_F091RC-1.7.0.2.zip...OK
Updating to 1.7.0.2

Connected to JTAG device with ID 0666FF484956827867164917

Device: STM32F09x/F030xC
Board: NUCLEO-F091RC
CPU: Cortex-M0
Device ID: 0x442

******************************************* WARNING ***************************************
It seems that the firmware image that's about to be used isn't the appropriate one for the
target connected. But this doesn't necessarily mean that it won't work.
*******************************************************************************************

Flashing device...
nanoBooter.hex
nanoCLR.hex
Flashing completed...
Reset MCU on device... OK
```

### Detailed repro steps so we can see the same problem

1. Installed the latest vs studio extension of nanoFramework
2. Installed STM32CubeProgrammer and performed a full chip erase
3. Flashed the device with `$ nanoff --update --target ST_NUCLEO64_F091RC --jtag`
4. The device still doesn't show up in VS

msschl, 2021-10-06:
Well, it seems like after rebooting my machine and flushing the STM like described in [flash-cube-programmer.html#flash-a-jtag-connected-board](https://docs.nanoframework.net/content/stm32/flash-cube-programmer.html#flash-a-jtag-connected-board) the device is found after step 7 of the linked article in the Device Explorer of VS2019.
However, I am still unable to deploy to the board. And step 8. of the article results in an error:

```txt
  16:10:47 : ST-LINK SN  : 0666FF484956827867164917
  16:10:47 : ST-LINK FW  : V2J29M18
  16:10:47 : Board       : NUCLEO-F091RC
  16:10:47 : Voltage     : 3.28V
  16:10:47 : SWD freq    : 4000 KHz
  16:10:47 : Connect mode: Normal
  16:10:47 : Reset mode  : Software reset
  16:10:47 : Device ID   : 0x442
  16:10:47 : Revision ID : Rev 1.0
  16:10:48 : UPLOADING OPTION BYTES DATA ...
  16:10:48 :   Bank          : 0x00
  16:10:48 :   Address       : 0x1ffff800
  16:10:48 :   Size          : 16 Bytes
  16:10:48 : UPLOADING ...
  16:10:48 :   Size          : 1024 Bytes
  16:10:48 :   Address       : 0x8000000
  16:10:48 : Read progress:
  16:10:48 : Data read successfully
  16:10:48 : Time elapsed during the read operation is: 00:00:00.007
  16:10:52 : MASS ERASE ... 
  16:10:52 : Mass erase successfully achieved
  16:10:52 : UPLOADING ...
  16:10:52 :   Size          : 1024 Bytes
  16:10:52 :   Address       : 0x8000000
  16:10:52 : Read progress:
  16:10:52 : Data read successfully
  16:10:52 : Time elapsed during the read operation is: 00:00:00.008
  16:11:00 : Memory Programming ...
  16:11:00 : Opening and parsing file: nanoBooter.hex
  16:11:00 :   File          : nanoBooter.hex
  16:11:00 :   Size          : 8940 Bytes
  16:11:00 :   Address       : 0x08000000 
  16:11:00 : Download in Progress:
  16:11:00 : File download complete
  16:11:00 : Time elapsed during download operation: 00:00:00.364
  16:11:00 : Verifying ...
  16:11:00 : Read progress:
  16:11:01 : Download verified successfully 
  16:11:01 : RUNNING Program ... 
  16:11:01 :   Address:      : 0x00000000
  16:11:01 : Application is running, Please Hold on...
  16:11:01 : Start operation achieved successfully
  16:22:32 : Memory Programming ...
  16:22:32 : Opening and parsing file: nanoCLR.hex
  16:22:32 :   File          : nanoCLR.hex
  16:22:32 :   Size          : 161276 Bytes
  16:22:32 :   Address       : 0x08002800 
  16:22:32 : Download in Progress:
  16:22:38 : File download complete
  16:22:38 : Time elapsed during download operation: 00:00:05.966
  16:22:38 : Verifying ...
  16:22:38 : Read progress:
  16:22:38 : Error: Data mismatch found at address  0x08002800 (byte = 0xFF instead of 0xC0)
  16:22:38 : Error: Download verification failed
  16:22:38 : RUNNING Program ... 
  16:22:38 :   Address:      : 0x00000000
  16:22:38 : Application is running, Please Hold on...
  16:22:38 : Start operation achieved successfully

```

msschl, 2021-10-06:
When trying to deploy using VS I get the following error:
`[*** ERROR: device reporting no assemblies loaded. This can not happen. Sanity check failed ***]`

Which seems to come from [DeployProvider.cs#L159](https://github.com/nanoframework/nf-Visual-Studio-extension/blob/4bedc9190370d26e73d70c936108d58334431340/VisualStudio.Extension/DeployProvider/DeployProvider.cs#L159).

networkfusion, 2021-10-06:
Hey @msschl , although this might indeed be an issue, it might be worth joining our discord community for support first, as you are likely to receive quicker support there... https://discord.gg/gCyBu8T

msschl, 2021-10-08:
Looks like the flash on my development board is faulty. I try getting a replacement board and test again.