nanoframework/Home #852: ESP32 native Capacitive Touch

alberk8, 2021-10-06:
**nanoFramework area:** (Visual Studio extension | Other tools | Hardware/target board | Community targets | Community contributions | Documentation )
Hardware: ESP32

### Is your feature request related to a problem? Please describe.
A c# class library to enable Capacitive Touch.
 
### Describe the solution you'd like
A native and C# class library to enable capacitive touch on ESP32.

### Describe alternatives you've considered
None.

closed 2022-04-16 by stale[bot]

BWalti, 2022-10-14:
I'd be interested in this too - but for now I found a work around using https://github.com/nanoframework/nanoFramework.IoT.Device/tree/develop/devices/Mpr121 - it would just be really nice, if one only needs a few Touch points to use the embedded functionality.

Ellerbach, 2022-10-14:
The thing is that there are quite some different drivers. They are similar but different. We have so far 3 available. They are mainly some from M5 Stack devices. See https://github.com/nanoframework/nanoFramework.IoT.Device/tree/develop/devices#touch-sensors

They can be used with any other devices. This is more flexible than having them in native. So it's on purpose that so far we don't have them in native. It really gives a lot of flexibility. They implement what they support. Some are single touch, some are double touch, some even support basic gesture but cheap copies does not for example. 

If we add them like for the big screen drivers in native, you'll have to build a specific image for each specific device using them. And on top, you'll anyway need a nuget (so far candidate would be nanoFramework.Graphics).

So it's not a work around, it's on purpose :-)

Ellerbach, 2022-10-14:
> @Ellerbach this is a _thing_ from the Xtensa MCUs that Espressif uses: some of their GPIOs can be configured to perform as touch sensors. Nothing related with UI and touch screens.

hoooooo, I miss understood it's "just" about the GPIO. Then, yes, for sure, as an extention, it can be a nice addition of course. Any doc somewhere on this?