nanoframework/Home #883: Phase Shifted Center Aligned?

jhancock4d, 2021-11-08:
Hi, I'm looking at controlling a half-bridge for DC/DC synchronous buck/boost.  I'm trying to figure out how to use the PWM code to get 2 pins, set pin 2 180 degrees offset from pin 1, center aligned and insert dead time so that pin 2 has "shoulders" from pin 1. so that shoot through doesn't occur on the mosfets.

How would one go about doing this? In Arduino code, we have to drop down to low level logic, and take 2 pins from the same timer, then set the pin to be inverted, and then add the dead time.  I can't find any example or even any indication that this is possible in nanoframework.

The same requirement occurs for all ESC motor control for BLDC, linear motors, actuators, and any 3 phase control, so I was hoping that there would just be some way to create some sort of paired PWM pin controller that would allow pulling 2 pints from the same timer and just took a dead time, and frequency and would fail if the hardware didn't support it, but otherwise would set it.  I.e. on an ESP32, the resolution decreases the faster you set the frequency, so it's a trade off. You have different trade offs on ATMega chips.

Suggestions on how to do this?  Is it even possible?

jhancock4d, 2021-11-08:
@josesimoes That explains it! Thank you!

networkfusion, 2021-12-08:
@jhancock4d did you find a workable solution for this?

jhancock4d, 2021-12-08:
@networkfusion I didn't. I suspect that this would require interop to make work, which means that a library that does this needs to be brought in and coded against. The FOC project for BLDC motors has the code, it's a matter of adapting it.  This project having ESC with FOC and Phase Shifting, and alignment on PWM would be key to further adoption because it's one of the major use cases for these microcontrollers.

networkfusion, 2021-12-08:
https://github.com/nanoframework/nanoFramework.IoT.Device/blob/79a350be3978312f9b8ed60679b0b6bddfda0a4b/devices/Uln2003/README.md

and

https://github.com/nanoframework/nanoFramework.IoT.Device/blob/79a350be3978312f9b8ed60679b0b6bddfda0a4b/devices/ServoMotor/README.md

may also help?

jhancock4d, 2021-12-09:
@networkfusion Sadly they don't.  Steppers don't use Phased PWM.  If there was BLDC control code then I could extract that or add to that to create a standard half-bridge. (in fact BLDC motor control uses half-bridges by definition)

Anyone have any BLDC motor control code? If not, it definitely should be added.  Basically this is what needs to be done: https://github.com/simplefoc/Arduino-FOC or mapped to, but the half-bridge needs to be broken out too instead of being hidden.

Ellerbach, 2021-12-09:
I guess this may not help niether? https://github.com/nanoframework/nanoFramework.IoT.Device/blob/develop/devices/DCMotor/README.md

Ellerbach, 2021-12-14:
So this would need to be done on the native side and exposed regarding the precision required. If you have to control in a very precise way 2 PWM I guess, there are not too many options. Still, the higher level can be exposed in a very simple way. Would you have some graph showing the states of both pins over the time?

josesimoes, 2021-12-14:
@jhancock4d is this being used (to be used) in a comercial application?

jhancock4d, 2021-12-14:
@josesimoes It isn't at the moment, but if I can prove it out, it will be going into a complete IoT infrastructure with various commercial components. For the moment, this is me testing theories.

I'm also suggesting that this is a good idea for you, because micro-controllers are disproportionately used for driving motors.

(and so would integration with ESPHome built in, but that's a different topic)

closed 2022-04-16 by stale[bot]