nanoframework/Home #885: HttpListener generating exceptions while getting messages

Ellerbach, 2021-11-14:
### Details about Problem

HTTP listener generates exceptions while used in a simple http port 80 mode.

**.NET nanoFramework library:** System.Net and System.Net.Http

**VS version:** (2017 | 2019 )

**VS extension version:**

**Target:**

ESP32 with or without SPRAM

**Device capabilities output:**

### Description

Run the basic HTTP Listener sample from the repo, adjust for port 80.

access the server with a browser just with the IP address.

You'll get exceptions:

```text
Listening for HTTP requests @ 192.168.1.137:80 ...
    ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (1) ++++
    ++++ Message: 
    ++++ System.Net.Sockets.NativeSocket::ioctl [IP: 0000] ++++
    ++++ System.Net.Sockets.Socket::get_Available [IP: 0018] ++++
    ++++ System.Net.Sockets.NetworkStream::get_Length [IP: 001f] ++++
    ++++ System.Net.InputNetworkStreamWrapper::Read_HTTP_Line [IP: 004b] ++++
    ++++ System.Net.HttpListenerRequest::ParseHTTPRequest [IP: 000d] ++++
    ++++ System.Net.HttpListenerContext::get_Response [IP: 000d] ++++
    ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (1) ++++
    ++++ Message: 
    ++++ System.Net.Sockets.NativeSocket::ioctl [IP: 0000] ++++
    ++++ System.Net.Sockets.Socket::get_Available [IP: 0018] ++++
    ++++ System.Net.Sockets.NetworkStream::get_Length [IP: 001f] ++++
    ++++ System.Net.InputNetworkStreamWrapper::Read_HTTP_Line [IP: 004b] ++++
    ++++ System.Net.HttpListenerRequest::ParseHTTPRequest [IP: 000d] ++++
    ++++ System.Net.HttpListenerContext::get_Response [IP: 000d] ++++
    ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (1) ++++
    ++++ Message: 
    ++++ System.Net.Sockets.NativeSocket::ioctl [IP: 0000] ++++
    ++++ System.Net.Sockets.Socket::get_Available [IP: 0018] ++++
    ++++ System.Net.Sockets.NetworkStream::get_Length [IP: 001f] ++++
    ++++ System.Net.InputNetworkStreamWrapper::Read_HTTP_Line [IP: 004b] ++++
    ++++ System.Net.HttpListenerRequest::ParseHTTPRequest [IP: 000d] ++++
    ++++ System.Net.HttpListenerContext::get_Response [IP: 000d] ++++
```

### Detailed repro steps so we can see the same problem

see above

### Other suggested things

### Expected behaviour

No exception at all should be raised

#### Sample Project

Use the HttpListener sample from the sample repository.

### Screenshot

### Additional context

This is blocking as well for WebServer and any scenario depending on it.
This seems to work before the migration to latest IDF for ESP32 target. Not tested on a STM32 device.

### I've made an effort to fix the bug

lafe, 2021-11-21:
Is the ID after the `CLR_E_Fail` relevant? I tried it with my ESP32 and the WebServer.GpioRest example and got IDs 5 and 6:
```
++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (6) ++++
++++ Message: 
++++ System.Net.Sockets.NativeSocket::ioctl [IP: 0000] ++++
++++ System.Net.Sockets.Socket::get_Available [IP: 0018] ++++
++++ System.Net.Sockets.NetworkStream::get_Length [IP: 001f] ++++
++++ System.Net.InputNetworkStreamWrapper::Read_HTTP_Line [IP: 004b] ++++
++++ System.Net.HttpListenerRequest::ParseHTTPRequest [IP: 000d] ++++
++++ System.Net.HttpListenerContext::get_Request [IP: 000d] ++++
++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (5) ++++
++++ Message: 
++++ System.Net.Sockets.NativeSocket::ioctl [IP: 0000] ++++
++++ System.Net.Sockets.Socket::get_Available [IP: 0018] ++++
++++ System.Net.Sockets.NetworkStream::get_Length [IP: 001f] ++++
++++ System.Net.InputNetworkStreamWrapper::Read_HTTP_Line [IP: 004b] ++++
++++ System.Net.HttpListenerRequest::ParseHTTPRequest [IP: 000d] ++++
++++ System.Net.HttpListenerContext::get_Request [IP: 000d] ++++
```

Ellerbach, 2021-11-22:
> Is the ID after the `CLR_E_Fail` relevant? I tried it with my ESP32 and the WebServer.GpioRest example and got IDs 5 and 6:

Not sure if it is. I know the investigation to fix all this is on @josesimoes radar. Thanks for the additional elements!

closed 2021-11-28 by josesimoes

belem2050, 2025-01-15:
Hey everyone!
Was that fixed? Because I am getting exactly the same issue and it is rebooting every single time one of those exceptions occurs, which is really critical for my project since I am storing some global states.

If anyone has a suggestion, I'd be glad to hear.

Thanks in advance.

CoryCharlton, 2025-01-15:
> Hey everyone!
> Was that fixed? Because I am getting exactly the same issue and it is rebooting every single time one of those exceptions occurs, which is really critical for my project since I am storing some global states.
> 
> If anyone has a suggestion, I'd be glad to hear.
> 
> Thanks in advance.

I'm using the `HttpListener` without any issues. You probably want to hop on the Discord and ask a question with more details or open a new issue without enough code to reproduce the problem.