nanoframework/Home #886: Add persistent storage using part of MCUs flash

thwr, 2021-11-15:
**nanoFramework area:** 
General feature for most targets

### Is your feature request related to a problem? Please describe.
Yes and no: Currently there is no way to save configuration parameters across restarts.

### Describe the solution you'd like
In embedded projects, you often need to store a very small but configurable set of configuration parameters. The configuration can be done via a UI (Display + Keys), via a network interface (REST API, ...) or even by using desired properties of Azure IoT Hub DeviceTwin, for example.

Regardless of how the configuration is done, there should be a way to have it persist between power loss / reboot events.

To achieve this goal, nanoFramework could provide frontend applications with memory backed up by a small portion of the respective MCU's flash memory (in the range of a few kilobytes). This memory does not need to be managed in any way, frontend applications could use (binary) serialization/deserialization to store and retrieve data, for example.

### Describe alternatives you've considered
External memory such as I2C/SPI EEPROMs, SD cards, etc. could suffice for this task, but require additional hardware, drivers, and application logic. On the other hand, nanoFramework already uses a system like the one described above to store network configuration parameters and certificates, so it should not be too complicated.

### Additional context
This feature should be beneficial for almost all projects that go beyond "hello world". So it would be nice to see this feature becoming reality.

### Submit a PR with the implementation
n/a

networkfusion, 2021-12-13:
@thwr Is there any reason why SPIFFS is not appropriate?

thwr, 2023-08-20:
> SPIFFS

Oh my... missed this one, sorry

Yes, because that requires an external IC that might not be available on off-the-shelve hardware. Already mentioned that in my initial feature request

alberk8, 2023-08-20:
> > SPIFFS
> 
> Oh my... missed this one, sorry
> 
> Yes, because that requires an external IC that might not be available on off-the-shelve hardware. Already mentioned that in my initial feature request

Yes, there is internal spiff storage in the ESP32 platform.   Sample [here](https://github.com/nanoframework/Samples/blob/main/samples/System.IO.FileSystem/BasicFileSystemExample/Program.cs), the I: drive is the internal storage.

thwr, 2023-08-20:
Perfectly possible, but isn't your approach platform specific?

alberk8, 2023-08-20:
Not every platform support partitioning of the flash as flexible and with large space as ESP32.

thwr, 2023-08-20:
Yeah, I wouldn't like to go down any kind of religious path here: Every MCU has it's own pro and con's. For example, for my needs, anything ESP-like (even the current generation) would be a poor choice.

Anyway, my suggested approach is platform independent. I would rather like to see that supported. Especially because it's already implemented for the networking configuration stuff which should be relatively easy to build upon.