nanoframework/Home #890: Need images for ESP32-S2 series without the internal USB CDC

josesimoes, 2021-11-18:
### Details about Problem

**Target:** ESP32-S2 

### Description

All current images for ESP32-S2 are using the internal USB CDC for Wire Protocol.
Several modules out there have an external serial-USB chip instead. Need to figure out a common pattern for the UART pins so we can offer an image that fits those.
Example: https://www.olimex.com/Products/IoT/ESP32-S2/ESP32-S2-DevKit-Lipo/open-source-hardware

hneel, 2022-02-23:
Does this apply to the Wemos S2 Mini too?  
I can't get a connection with this device

```
C:\>nanoff --serialport COM12 --target ESP32_PSRAM_REV0 --update
.NET nanoFramework Firmware Flasher v2.0.30+937d96cbf5
Copyright (C) 2019 .NET Foundation and nanoFramework project contributors

Using COM12 @ 1500000 baud to connect to ESP32.
Reading details from chip...
Error E4000: Error executing esptool command. (esptool.py v3.2
Serial port COM12
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...

Traceback (most recent call last):
  File "esptool.py", line 377, in detect_chip
struct.error: unpack requires a buffer of 16 bytes

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "esptool.py", line 5143, in <module>
  File "esptool.py", line 5136, in _main
  File "esptool.py", line 4541, in main
  File "esptool.py", line 121, in get_default_connected_device
  File "esptool.py", line 388, in detect_chip
  File "esptool.py", line 663, in connect
  File "esptool.py", line 616, in _connect_attempt
  File "serial\serialutil.py", line 594, in inWaiting
  File "serial\serialwin32.py", line 259, in in_waiting
serial.serialutil.SerialException: ClearCommError failed (PermissionError(13, 'The device does not recognize the command.', None, 22))
[3356] Failed to execute script 'esptool' due to unhandled exception!
)

```

josesimoes, 2022-03-02:
@hneel I've check the Wemos S2 Mini schematic and it's using the embedded USB CDC port, so this issue does not apply to that hardware. 

Instead of the above, try this command line: `nanoff --serialport COM12 --platfomr esp32 --update --preview`.

josesimoes, 2022-03-02:
@hneel can you go over to our Discord to discuss about this? If yes, please ping me there, about this.

josesimoes, 2022-03-08:
For reference, [here's](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-saola-1-v1.2.html#hardware-reference) an official module using an external USB-Serial bridge chip.

FrankBakkerNl, 2023-05-13:
As discussed on Discord this board seems to have the same issue
 http://www.lilygo.cn/prod_view.aspx?typeid=50033&id=1321&fid=t3:50033:3

networkfusion, 2024-12-16:
@AdrianSoundy Is this now fixed?

AdrianSoundy, 2024-12-17:
I haven't looked at this one but it is straight forward.
We currently have 2 firmware for S2
- Kaluga  with Uart for wire protocol & screen ILI9341 
- Feather_S2 with USB CDC for wire protocol 

The Kaluga  should work for all devices with just UART

Probably best to create 2 new firmware and retire these one.
ESP32_S2_UART
ESP32_S2_USB
Both with generic screen drivers

What do you think ?
@josesimoes @networkfusion

josesimoes, 2024-12-17:
I'm OK with that option. Will you tackle that?

closed 2024-12-19 by AdrianSoundy