nanoframework/Home #892: Plain Old Class Object not returning Unique Hash Code

alberk8, 2021-11-19:
### Details about Problem
Plain Old Class Objects (POCO) are not returning unique hash code. 

**Target:**
ESP32

**Firmware image version:**
1.7.1-preview.1125

Worked before? If so, with which nanoFramework image version:
First time encountering the issue.

### Description
The POCO are returning zero or negative int hash code.   Also when multiple instance are created.

### Detailed repro steps so we can see the same problem

```
public class Program
{
	public static void Main()
	{
		var a = new TestClass();
		var b = new TestClass();
		
		Debug.WriteLine($"HashCode A {a.GetHashCode()} HashCode B {b.GetHashCode()}");
		
		var c = new NewClass();
		var d = new NewClass();
		
		Debug.WriteLine($"HashCode C {c.GetHashCode()} HashCode D {d.GetHashCode()}");
	}
	
}

```
```
 public class TestClass
    {
        private int _int;
        public TestClass()
        {
            Debug.WriteLine($"TestClass Constructor HashCode {GetHashCode()}");
            // if _int is moved aboved the Debug line then the GetHashCode will return Value
            _int = 100;
        }

        public void DoSomething()
        {
            _int = 1 + 2;
        }
    }

    public class NewClass
    {
        public string Name { get; set; }
        public string Address { get; set; }
    }
```

### Other suggested things

### Expected behaviour
Expect to get unique hash code for every instance of the class and do not return zero value..

### Additional context

Add any other context about the problem here.
```
\\ Debug Output
TestClass Constructor HashCode 0
TestClass Constructor HashCode 0
HashCode A 1192204961 HashCode B 1192204961
HashCode C 0 HashCode D 0
```

### Make an effort to fix the bug
Attempted but could not get to the root cause.

alberk8, 2021-11-20:
After further testing.  I manage to find two patterns for Hash code == 0.   This is when property where get and set are defined without any logic or private fields
```
private int _int;
// or
public string Name { get; set; }

```

josesimoes, 2021-11-22:
Cross check for .NET 6.0: https://dotnetfiddle.net/FWy25q

closed 2021-11-22 by josesimoes