nanoframework/Home #896: Wifi automatically connected on subsequent run even without initialization

alberk8, 2021-11-24:
### Details about Problem
Wifi STA mode is connected to Access Point even when it is commented out in code.

**Target:**
ESP32 WROVER

**Firmware image version:**
1.7.1-preview.1143

Worked before? If so, with which nanoFramework image version:

### Description
After initial successful WIFI connection to AP the ESP32 will still connect to the WIFI even when code is commented up in subsequent deployment.

### Detailed repro steps so we can see the same problem
1) run the code below.
2) comment out the "bool success" line and redeploy.   
3) you will see the IP of the ESP32, gateway address and if you goto the AP UI, you will see the ESP32 connected.

```
 CancellationTokenSource cs = new(60_000);
 bool success = NetworkHelper.ConnectWifiDhcp(c_SSID, c_AP_PASSWORD,
                reconnectionKind: Windows.Devices.WiFi.WiFiReconnectionKind.Automatic, setDateTime: true, token: cs.Token);

foreach (var net in NetworkInterface.GetAllNetworkInterfaces())
{
      Debug.WriteLine($"NetworkType: {net.NetworkInterfaceType}  IP Address: {net.IPv4Address}");
      Debug.WriteLine($"Gateway Address : {net.IPv4GatewayAddress} ");
}
```

### Other suggested things

### Expected behaviour
Wifi is not automatically activated if there is no code to activate it even if prior run has successfully activated WIFI.

### Screenshot
None

### Additional context

### Make an effort to fix the bug
None

AdrianSoundy, 2021-12-12:
The default behavior for ESP32 is for the WiFi to automatically connect if it has saved WiFi credentials. If you don't want it to connect you have to not save credentials.

alberk8, 2021-12-12:
> The default behavior for ESP32 is for the WiFi to automatically connect if it has saved WiFi credentials. If you don't want it to connect you have to not save credentials.

How do i not save the automatic reconnect without save credentials?

closed 2022-04-16 by stale[bot]