nanoframework/Home #897: Non TLS HTTP calls fails

alberk8, 2021-11-25:
### Details about Problem
plain HTTP (non TLS) call/connection to website failed with error.

**.NET nanoFramework library:** (name of the library)
nanoframework/System.Net.Http

**VS version:** (2017 | 2019 )
2019 Version 16.11.3

**VS extension version:**
2019.8.0.19

**Target:**
ESP32 WROVER

**Device capabilities output:**
1.7.1-Preivew.1144

### Description
Trying out the example on httpwebrequest from the nanoframework/samples/Http, with that example which uses TLS it is successful but when I change to http (http://example.com/) only (commenting out the TLS and Cert option).  I get this error.
```
Performing Http request to: http://example.com
    ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (1) ++++
    ++++ Message: 
    ++++ System.Net.Sockets.Socket::get_Available [IP: 0018] ++++
    ++++ System.Net.Sockets.NativeSocket::ioctl [IP: 0000] ++++
    ++++ System.Net.Sockets.NetworkStream::get_Length [IP: 001f] ++++
    ++++ System.Net.InputNetworkStreamWrapper::Read_HTTP_Line [IP: 004b] ++++
    ++++ System.Net.HttpWebRequest::ParseHTTPResponse [IP: 002e] ++++
    ++++ System.Net.HttpWebRequest::GetResponse [IP: 0040] ++++
    ++++ HttpRequest.Program::Main [IP: 009e] ++++
    ++++ Exception System.Net.WebException - 0x00000000 (1) ++++
    ++++ Message: GetResponse() failed
    ++++ HttpRequest.Program::Main [IP: 009e] ++++
    ++++ System.Net.HttpWebRequest::GetResponse [IP: 00dd] ++++
Exception thrown: 'System.Net.WebException' in System.Net.Http.dll
```
### Detailed repro steps so we can see the same problem

1. Use this sample.  https://github.com/nanoframework/Samples/tree/main/samples/HTTP/HttpWebRequest

2. change the following lines
```
string url = "http://example.com";
//httpWebRequest.SslProtocols = System.Net.Security.SslProtocols.Tls12;
//httpWebRequest.HttpsAuthentCert = rootCACert;
```

...

### Other suggested things

### Expected behaviour
The webrequest should be successful.

#### Sample Project

### Screenshot

### Additional context

### I've made an effort to fix the bug
Trace the error to socket where it is create but throws error when trying to access socket.Available.
[https://github.com/nanoframework/System.Net.Http/blob/0591afcb6205ef09a2eaed7b5e9ffdd768a85598/nanoFramework.System.Net.Http/Http/System.Net.HttpWebRequest.cs#L1469](url)

alberk8, 2021-11-26:
I manage to track down the native code that is causing the error but it is beyond my expertise to attempt to fix it.

[https://github.com/nanoframework/nf-interpreter/blob/b60a6d6ccda11a0a64c81a3ab00f5adb5de25219/src/DeviceInterfaces/System.Net/sys_net_native_System_Net_Sockets_NativeSocket.cpp#L423](url)

alberk8, 2021-11-28:
After some more debugging.  I have locate the issue down to native IOCTL where LWIP stack needs the "#define LWIP_SO_RCVBUF" to be enabled.

On the ESP32 there is a corresponding config "CONFIG_LWIP_SO_RCVBUF=y" needs to be set but it seems that setting this has no effect.

I am on the Dev Container and I did the mandatory deleting the build directory after changing the "sdkconfig.default.xxx" file.

But the build does not enable the LWIP_SO_RCVBUF.   What I did to do testing was manually set the "#define" below of the "targets/ESP32/_include/lwipopts.h".

[https://github.com/nanoframework/nf-interpreter/blob/cf22b148ccdda5d96e8ef8fe8ffc5939b0ac7109/targets/ESP32/_Include/lwipopts.h#L610](url)  

#define LWIP_SO_RCVBUF         1   //CONFIG_LWIP_SO_RCVBUF

I don't know where else to set to get ESP32 sdkconfig to work.  The other platform also needs looking into.

AdrianSoundy, 2021-11-28:
@alberk8 
Good for getting so far. You have changed the right parameter. I assume this is ESP32_rev0 build.
When you delete the build directory also delete the sdkconfig in the root directory.
There may be a problem with it not being refreshed if it already exists.  I hit a similar problem recently but hadn't investigated cause.

lafe, 2021-11-28:
Great analysis, @alberk8. This might also be the same underlying issue that causes #885 and #882.

josesimoes, 2021-11-28:
This prompted a deep analysis on all lwIP options. Just found out that there are others options that are either misaligned with the expected configs for System.Net and networking and even others that are enabled needlessly.

lafe, 2021-11-28:
The change proposed by @alberk8 works for issue #885. I deleted the skdconfig and the build before compiling the custom firmware and afterwards everything works (after figuring out how to flash the image - PR incoming for the documentation...)

closed 2021-11-28 by josesimoes