nanoframework/Home #898: VSCode extension: can't run command from Command Palette with message "command xxx not found"

LiamSho, 2021-11-26:
### Details about Problem

**.NET nanoFramework area:** Visual Studio Code extension

**VSCode version:** v1.62.3-Universal (MacBook Air M1)

**VSCode extension version:** v1.0.15-preview

### Description

Can not run any nanoFramework command from VSCode command palette.

### Detailed repro steps so we can see the same problem

1. Open Visual Studio Code with .NET nanoFramework extension installed.

2. Open a folder which contains the nanoFramework project. (I use the blinky sample in nanoframework/Sample repo)

3. Open Command Palette and run `nanoFramework: Build Project`

4. A dialog appeared with message: `Command 'nanoFramework: Build Project' resulted in an error (command 'vscode-nanoframework.nfbuild' not found)`  and the command failed to run.

### Other suggested things

This is my working environment. I don't know if they are helpful or not. But I hope these thing will be helpful :)

I am working on a MacBook Air with M1 chip (macOS 12.0.1). I have installed .NET 5 (x64) and .NET (x64 and arm64). The arm version of .NET is install in `/usr/local/share/dotnet` and the x64 version of .NET is installed in `/usr/local/share/dotnet/x64`. I made an alias in my .zshrc file to run `/usr/local/share/dotnet/x64/dotnet` with a simple `dotnet64` command. Because nanoff require .NET 5 to run, I installed it with command `dotnet64 tool install nanoff -g`. I have also installed Mono(x64, v6.12.0.122-stable).

### Expected behaviour

The `Command 'nanoFramework: Build Project' resulted in an error (command 'vscode-nanoframework.nfbuild' not found)`  message shouldn't appear and the command should run.

### Screenshot

[image]

josesimoes, 2021-11-26:
This is an issue with the extension packaging. It's being investigated.

DocBrown101, 2021-11-26:
FYI: The same happens with an Ubuntu operating system.

josesimoes, 2021-12-07:
We finally got to the bottom of this. It's fixed with nanoframework/nf-VSCodeExtension#11.
Version 1.0.30 it's working as expected.

closed 2021-12-07 by josesimoes