nanoframework/Home #901: Icons not visible in VS extension 2022.0.3.12

oblaise, 2021-12-01:
### Details about Problem

**.NET nanoFramework area:**  Visual Studio extension

**VS version: 2022 - 17.0.1**

**VS extension version:** 2022.0.3.12

### Description

I've removed and reinstalled today the VS extension as it was recommended in your tweet of Yesterday. The icons aren't visible anymore (in the 'other windows' menu and in the extension window itself).

### Detailed repro steps so we can see the same problem
1. Uninstall previous version (All was OK)
2. Reinstall current version

### Screenshot
[image]

### Additional context

### Make an effort to fix the bug

Ellerbach, 2021-12-01:
I think this may help:
[image]

oblaise, 2021-12-01:
It was exactly what I did : "I've removed and reinstalled today the VS extension as it was recommended in your tweet of Yesterday. The icons aren't visible anymore (in the 'other windows' menu and in the extension window itself)."

Ellerbach, 2021-12-01:
> It was exactly what I did : "I've removed and reinstalled today the VS extension as it was recommended in your tweet of Yesterday. The icons aren't visible anymore (in the 'other windows' menu and in the extension window itself)."
> 
> Before that the installation was working correctly.

I reproduced as well. I rolled back to a very old version.

josesimoes, 2021-12-01:
Is there any difference if you uninstall, reboot and install from scratch?

oblaise, 2021-12-01:
I just tried but this didn't solve the issue.

closed 2021-12-02 by josesimoes

josesimoes, 2021-12-02:
@oblaise can you please install v2022.0.3.14 from OpenVsix gallery and confirm that you're seeing the buttons now?

Note that if you had the Device Explorer window open the last time you've closed VS, you may get an error message about the previous instance of the tool window. After opening the "new" one from the VS other windows meu, everything should be OK.

oblaise, 2021-12-02:
It's OK with v2022.0.3.14 icons are visible in the menu and in the panel. Txs.

josesimoes, 2021-12-03:
Aditional information about this matter: it seems to be related with VS image cache.
Following this question [here](https://developercommunity2.visualstudio.com/t/Some-icons-are-missing-in-the-VS-toolbar/549394?entry=problem&ref=native&refTime=1638546032559&refUserId=b13b3249-a663-6ddd-b0fd-d39c00e98eca) and after deleting the image cache and running `devenv /updateConfiguration`  from VS developer command prompt, the icons seems to be showing correctly.

networkfusion, 2021-12-08:
Without deleting the image cache and just running `devenv /updateConfiguration` from the developer command prompt is sufficient. This should be added to announcements or run as part of the VS extension install (IMHO)...