nanoframework/Home #903: nanoFramework.Logging not working in unit tests

Imposter, 2021-12-04:
### Details about Problem

**.NET nanoFramework area:** VS Extension/Unit Test Runner

**VS version:** 2019

**VS extension version:** 2019.10.0.2

**Target:** ESP32

**Firmware image version:** Unknown

**Device capabilities output:** N/A

### Description

I tried to create a unit test for a library which references (and uses) the nanoFramework.Logging library. There seems to be some sort of a version mismatch (to mscorlib) and due to that, references are not resolved when running the unit test.

Here's a minimal sample showing the issue: https://github.com/Imposter/nf-logging-unittests

### Detailed repro steps so we can see the same problem

1. Create library that references nanoFramework.Logging

2. Write unit test for some code in that library

### Expected behaviour

The unit test will run and succeed.

### Additional context

```
DoSomethingTest
Source: ExampleClassTests.cs line 17
Duration: < 1 ms

Message: 

.NET nanoFramework nanoCLR WIN32 v1.7.3-preview.3+c2e0748291

Copyright (c) 2020 .NET Foundation and Contributors

Starting...
Created EE.
Started Hardware.
Loading Assemblies.
  Assembly: ExampleLibrary (1.0.0.0)   (200 RAM - 516 ROM - 163 METADATA)

  AssemblyRef    =        8 bytes (       2 elements)
  TypeRef        =       16 bytes (       4 elements)
  FieldRef       =        0 bytes (       0 elements)
  MethodRef      =       12 bytes (       3 elements)
  TypeDef        =        8 bytes (       1 elements)
  FieldDef       =        4 bytes (       1 elements)
  MethodDef      =        4 bytes (       2 elements)
  StaticFields   =        0 bytes (       0 elements)

  Attributes      =        0 bytes (       0 elements)
  TypeSpec        =        0 bytes (       0 elements)
  Resources       =        0 bytes (       0 elements)
  Resources Files =        0 bytes (       0 elements)
  Resources Data  =        0 bytes
  Strings         =      196 bytes
  Signatures      =       19 bytes
  ByteCode        =       32 bytes

  Assembly: NFUnitTest (1.0.0.0)   (244 RAM - 784 ROM - 282 METADATA)

  AssemblyRef    =       16 bytes (       4 elements)
  TypeRef        =       32 bytes (       8 elements)
  FieldRef       =        0 bytes (       0 elements)
  MethodRef      =       32 bytes (       8 elements)
  TypeDef        =        8 bytes (       1 elements)
  FieldDef       =        0 bytes (       0 elements)
  MethodDef      =        8 bytes (       3 elements)
  StaticFields   =        0 bytes (       0 elements)

  Attributes      =       24 bytes (       3 elements)
  TypeSpec        =        0 bytes (       0 elements)
  Resources       =        0 bytes (       0 elements)
  Resources Files =        0 bytes (       0 elements)
  Resources Data  =        0 bytes
  Strings         =      345 bytes
  Signatures      =       10 bytes
  ByteCode        =       26 bytes

  Assembly: mscorlib (1.11.7.2)   (4028 RAM - 31744 ROM - 19133 METADATA)

  AssemblyRef    =        0 bytes (       0 elements)
  TypeRef        =        0 bytes (       0 elements)
  FieldRef       =        0 bytes (       0 elements)
  MethodRef      =        0 bytes (       0 elements)
  TypeDef        =     1152 bytes (     144 elements)
  FieldDef       =      192 bytes (      96 elements)
  MethodDef      =     1596 bytes (     798 elements)
  StaticFields   =      144 bytes (      12 elements)

  Attributes      =       40 bytes (       5 elements)
  TypeSpec        =        4 bytes (       1 elements)
  Resources       =        0 bytes (       0 elements)
  Resources Files =        0 bytes (       0 elements)
  Resources Data  =        0 bytes
  Strings         =     2584 bytes
  Signatures      =     2097 bytes
  ByteCode        =     9898 bytes

  Assembly: nanoFramework.Logging (1.0.0.0)   (612 RAM - 3744 ROM - 2184 METADATA)

  AssemblyRef    =        4 bytes (       1 elements)
  TypeRef        =       68 bytes (      17 elements)
  FieldRef       =        0 bytes (       0 elements)
  MethodRef      =       48 bytes (      12 elements)
  TypeDef        =      104 bytes (      13 elements)
  FieldDef       =       16 bytes (       8 elements)
  MethodDef      =      128 bytes (      64 elements)
  StaticFields   =       36 bytes (       3 elements)

  Attributes      =      280 bytes (      35 elements)
  TypeSpec        =        0 bytes (       0 elements)
  Resources       =        0 bytes (       0 elements)
  Resources Files =        0 bytes (       0 elements)
  Resources Data  =        0 bytes
  Strings         =      781 bytes
  Signatures      =      260 bytes
  ByteCode        =      652 bytes

  Assembly: nanoFramework.TestFramework (1.0.166.0)   (632 RAM - 5520 ROM - 1898 METADATA)

  AssemblyRef    =        4 bytes (       1 elements)
  TypeRef        =       96 bytes (      24 elements)
  FieldRef       =        0 bytes (       0 elements)
  MethodRef      =      112 bytes (      28 elements)
  TypeDef        =       80 bytes (      10 elements)
  FieldDef       =        4 bytes (       1 elements)
  MethodDef      =      120 bytes (      60 elements)
  StaticFields   =       12 bytes (       1 elements)

  Attributes      =       16 bytes (       2 elements)
  TypeSpec        =        0 bytes (       0 elements)
  Resources       =        0 bytes (       0 elements)
  Resources Files =        0 bytes (       0 elements)
  Resources Data  =        0 bytes
  Strings         =     1148 bytes
  Signatures      =      246 bytes
  ByteCode        =     2346 bytes

  Assembly: nanoFramework.UnitTestLauncher (0.0.0.0)   (304 RAM - 1288 ROM - 481 METADATA)

  AssemblyRef    =        8 bytes (       2 elements)
  TypeRef        =       68 bytes (      17 elements)
  FieldRef       =        0 bytes (       0 elements)
  MethodRef      =       64 bytes (      16 elements)
  TypeDef        =        8 bytes (       1 elements)
  FieldDef       =        0 bytes (       0 elements)
  MethodDef      =        8 bytes (       3 elements)
  StaticFields   =        0 bytes (       0 elements)

  Attributes      =        0 bytes (       0 elements)
  TypeSpec        =        0 bytes (       0 elements)
  Resources       =        0 bytes (       0 elements)
  Resources Files =        0 bytes (       0 elements)
  Resources Data  =        0 bytes
  Strings         =      282 bytes
  Signatures      =      121 bytes
  ByteCode        =      395 bytes

Resolving.
Link failure: some assembly references cannot be resolved!!

Assembly: ExampleLibrary (1.0.0.0) needs assembly 'nanoFramework.Logging' (1.0.0.0)
Assembly: NFUnitTest (1.0.0.0) needs assembly 'ExampleLibrary' (1.0.0.0)
Assembly: NFUnitTest (1.0.0.0) needs assembly 'nanoFramework.Logging' (1.0.0.0)
Assembly: nanoFramework.Logging (1.0.0.0) needs assembly 'mscorlib' (1.10.5.4)
Error: a3000000
Exiting.
```

josesimoes, 2021-12-05:
The release for this library was missing. Just released and published.
Unit Tests running fine in the pipeline. Please check:
https://github.com/nanoframework/nanoFramework.Logging/commit/36e0efd9446a332d6f045290ad5e9db84cc333ce
and
https://dev.azure.com/nanoframework/nanoframework.Logging/_build/results?buildId=25340&view=results

Imposter, 2021-12-05:
@josesimoes Thanks for the quick update. I just updated NuGet packages and tried it and it works now. 👍

closed 2021-12-05 by josesimoes