nanoframework/Home #906: Web Server/HttpListener crahses ESP32

lafe, 2021-12-05:
### Details about Problem

**Target:** ESP32

**Firmware image version:** 1.7.3-preview.7 (1.7.3.7)

Worked before? If so, with which nanoFramework image version: -

**Device capabilities output:**

```
System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF v4.3.1
  Target:   ESP32
  Platform: ESP32

Firmware build Info:
  Date:        Dec  5 2021
  Type:        MinSizeRel build, chip rev. >= 0, without support for PSRAM
  CLR Version: 1.7.3.7
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  mscorlib, 1.11.7.2
  nanoFramework.Hardware.Esp32, 1.3.4.3
  nanoFramework.Hardware.Esp32.Rmt, 1.2.1.0
  nanoFramework.Networking.Sntp, 1.5.7.3
  nanoFramework.ResourceManager, 1.1.3.3
  nanoFramework.Runtime.Events, 1.9.2.3
  nanoFramework.Runtime.Native, 1.5.2.3
  nanoFramework.System.Collections, 1.3.0.3
  nanoFramework.System.Text, 1.1.2.3
  nanoFramework.WebServer, 1.0.0.0
  System.Device.Gpio, 1.0.2.3
  System.Device.Spi, 1.0.2.3
  System.IO.FileSystem, 1.0.2.0
  System.IO.Ports, 1.0.2.3
  System.Net, 1.7.1.3
  System.Net.Http, 1.3.6.2
  Windows.Devices.Wifi, 1.3.4.0
  Windows.Storage, 1.4.5.0
  Windows.Storage.Streams, 1.12.3.3
  Krippe, 1.0.0.0

Native Assemblies:
  mscorlib v100.5.0.14, checksum 0xC5322585
  nanoFramework.Runtime.Native v100.0.8.0, checksum 0x2307A8F3
  nanoFramework.Hardware.Esp32 v100.0.7.3, checksum 0xBE7FF253
  nanoFramework.Hardware.Esp32.Rmt v100.0.3.0, checksum 0x9A53BB44
  nanoFramework.Devices.OneWire v100.0.3.4, checksum 0xA5C172BD
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.0.1, checksum 0x5A31313D
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x210D05B1
[... 141 lines cut ...]

### Detailed repro steps so we can see the same problem

The problem is that this error cannot reproduced reliably. Sometimes, the request works flawlessly (even multiple times) but most of the time it fails. I already disconnected every external device and also tried a different ESP32 to make sure that there is no hardware defect. I also tried to use the context directly to build the response manually. However, the error is the same every time.

### Other suggested things
<!-- if applicable/relevant -->

### Expected behaviour

The response will be processed correctly and does not cause a reboot of the ESP32. It would also be great if the try...catch handler works correctly and catches these errors. 

### Screenshot
<!-- if applicable/relevant -->
<!--Very helpful if you send along a few screenshots to help visualize the issue!-->

### Additional context

-

### Make an effort to fix the bug

-

<!-- bug-report-fw-tag DO NOT REMOVE -->

josesimoes, 2021-12-05:
Please add here a minimal project reproducing this.

lafe, 2021-12-18:
Just a quick update: I was not yet able to build a minimal project due to time constraints. Hopefully, I will be able to investigate this issue further after Christmas.

josesimoes, 2022-01-18:
@lafe can you try with the latest NuGets? There were some improvements and might have fixed the issues you're experiencing.

josesimoes, 2022-01-20:
Closing this now. Please reopen if this is still an issue.

closed 2022-01-20 by josesimoes