nanoframework/Home #908: Remove bundled nanoff in VSCode extension and use system install of nanoff tool

LiamSho, 2021-12-08:
**nanoFramework area:** Visual Studio Code Extension

### Problem

I am working on a M1 chip powered MacBook. From .NET 6, `dotnet` command will run `arm64(native)` version of .NET installed in `/use/local/share/dotnet/dotnet`, To run some thing with `x64(rosseta)`, you need to run `/use/local/share/dotnet/x64/dotnet`. So I make an alias and use `dotnetx64` to run `/use/local/share/dotnet/x64/dotnet` and install nanoff global tool with this command and it works.

When deploy bin file to device. The extension will try to run a command like this:

``` sh
dotnet /Users/liamsho/.vscode/extensions/nanoframework.vscode-nanoframework-1.0.32/dist/utils/nanoFirmwareFlasher/nanoff.dll --target ESP32_REV0 --serialport /dev/cu.usbserial-0001 --deploy --image /Users/liamsho/Documents/Program/MCU_Course_Project/OutputDir/deploy.bin
```

In my case it will just throw an error like this:

`The framework 'Microsoft.NETCore.App', version '5.0.0' (arm64) was not found.`

.NET 5 do not have a amr64 version and I only install .NET 6 in arm64 but .NET 5 and 6 in x64.

I have two ways to fix this problem.

1. Change `dotnet` to `dotnetx64` or `/use/local/share/dotnet/x64/dotnet`
2. Change `dotnet /PATH_TO_PLUGIN/nanoff.dll ` to `nanoff`

###  Solution

In my opinion, just remove the bundled nanoff and use the system install of nanoff tool is OK. The size of nanoff tool is pretty large, and most of developers will have a global install of it.

It will resolve my problem and also make the size of the plugin much smaller.

bart-jansen, 2021-12-08:
I think as a quick win we could add some logic to use `nanoff` if the command exists, and use the bundled nanoff if it doesn't.

So something like:
```
if command -v nanoff &> /dev/null
then
     nanoff --target ESP32_REV0 --serialport /dev/cu.usbserial-0001 ...
else
    dotnet /Users/USER/.vscode/extensions/nanoframework.vscode-nanoframework-1.0.32/dist/utils/nanoFirmwareFlasher/nanoff.dll --target ESP32_REV0 --serialport /dev/cu.usbserial-0001 ..
fi
```

Thoughts?

josesimoes, 2021-12-08:
Can we "bundle" the install off nanoff dotnet tool in the extension? Like running the install command as one of the install steps, or at launch?

bart-jansen, 2021-12-08:
Currently, the extension already comes bundled with nanoff. Looks like the issue is not related to nanoff, but more related to the way we're invoking it.

Ellerbach, 2021-12-08:
> I think as a quick win we could add some logic to use `nanoff` if the command exists, and use the bundled nanoff if it doesn't.

I think that makes sense and will make things transparent and easy. 

> Can we "bundle" the install off nanoff dotnet tool in the extension? Like running the install command as one of the install steps, or at launch?

That's also an option. Now, when all is in the extension, it's quite straight forward and no brainer.