nanoframework/Home #912: Add mDNS service discovery to nanoframework

cristian-recoseanu, 2021-12-13:
**nanoFramework area:** (Hardware/target board)

### Is your feature request related to a problem? Please describe.
I believe adding mDNS service discovery support to nanoframework will help the community tackle more zero configuration use cases.

### Describe the solution you'd like
I believe the minimum viable feature set is to be able to query and resolve services via mDNS.

Publishing mDNS services from nanoframework is probably only useful in a limited number scenarios like a pure peer to peer environment where there is no central hub. This scenario also probably cannot allow devices to go to sleep which limits its utility even further.

### Describe alternatives you've considered
Espressif already have support for this in their arduino libraries:
https://github.com/espressif/arduino-esp32/blob/a59eafbc9dfa3ce818c110f996eebf68d755be24/libraries/ESPmDNS/examples/mDNS-SD_Extended/mDNS-SD_Extended.ino

In terms of other alternatives I can't think of anything else which would fit zero configuration scenarios besides just sending raw UDP multicast packets but that just feels like re-inventing the wheel which led to mDNS existing in the first place.

### Additional context
This came as an initial question about existing mDNS support on Stackoverflow here:
https://stackoverflow.com/questions/70282102/are-there-any-means-currently-for-performing-service-discovery-mdns-or-unicast

It would also be great to gauge the community interest in this area.

networkfusion, 2021-12-31:
It might be worth considering porting something like this if possible: https://github.com/karlredgate/mDNS-sharp

josesimoes, 2022-01-03:
Having that as a manged alternative would probably be the best option.

lwIP and Azure RTOS NetX Duo have native support for it and we can enable that, but we haven't seen much (if any) request for this feature.

closed 2022-04-28 by stale[bot]

GoGoGadgetCasts, 2023-07-02:
Would be great to have mDNS in nanoFramework! I see the Twitter poll showed 96% in favour of adding: https://twitter.com/nanoFramework/status/1477998595404144648

For a specific use case, the popular open-source LED library, WLED, supports discovery of WLED instances via mDNS.

josesimoes, 2023-07-03:
Hey @GoGoGadgetCasts ! 😄 
Correct! We have this on the todo list but with low priority.
To add context: that pool was answered by 26 persons... hardly a significant sample of the vast nanoFramework community... 😉

Ellerbach, 2023-07-04:
Just as a note, it's really easy to implement. See the page on Wikipedia: https://en.wikipedia.org/wiki/Multicast_DNS.
And also the spec: https://datatracker.ietf.org/doc/html/rfc6762
But you have to use only local addresses like 254.x.x.x, if not, mDNS doesn't seem to be useful. So when in a normal network with a proper DHCP server, this feature is not useful. IT's only in the case you have no network configured and use only the automatic default addressing.
Did I miss something?

alberk8, 2023-07-04:
It is useful in a network where DHCP is enabled.  I basically allows for local domain name searches on network that does not have a configurable DNS server, essentially a home router.

Ellerbach, 2023-07-04:
ok, get it. So looking at .NET implementations, I found this one using an MIT license: https://github.com/richardschneider/net-mdns
I guess it's something that can be easily ported to nanoFramework?

josesimoes, 2023-07-04:
Worth nothing that ESP32 IDF has this component. Just like lwIP and Azure Netx have it, as I've mentioned earlier.

Not sure if, resource wise, this should be implemented as part of the network layer or enterily as a managed library...

networkfusion, 2023-07-04:
I was going to point out that we just need to enable MDNS in lwip and have a managed interface. See https://www.nongnu.org/lwip/2_0_x/group__mdns.html

alberk8, 2023-07-04:
> Worth nothing that ESP32 IDF has this component. Just like lwIP and Azure Netx have it, as I've mentioned earlier.
> 
> Not sure if, resource wise, this should be implemented as part of the network layer or enterily as a managed library...

I would prefer it to be in native.

xristiaan, 2023-11-03:
I jsut started using NanoFramework to build some ESP32 based home automation devices and would like to be able to add these as easily as possible to my Homey home automation hub. It's easy to create device specific drivers on Homey using mDNS-SD because Homey has build in support for that. So this would be a very usefull feature for me.

Ellerbach, 2023-11-07:
Yes, core idea is to surface in a nice way the mDNS from the natice lwip implementation. Depending on the needs, it will be quite straight forward. Still, someone needs to surface this.
I've started too at all this but so far, I did not have the proper time to start the implementation at all. First platform will definitely be ESP32.
As the manage API will be simple, this can even seats in the Networking nuget as in all the cases, a native implementation exists on all the MCU.