nanoframework/Home #918: Invalid exception handler execution flow

aromaa, 2021-12-23:
### Details about Problem

**Target:** ESP32-WROOM-32

**Firmware image version:** 1.7.3-preview.67

### Description
Exception flow is not correctly handled inside finally block when another method catches exception. This also required branching inside the try block so this is some weird edge case here. From the following code we can observe that the program prints out "Impossible!?!" even though the code is unreachable.
```csharp
try
{
	static bool False() => false;

	if (!False())
	{
		Debug.WriteLine("Return!");

		return;
	}
}
finally
{
	Throw();

	static void Throw()
	{
		try
		{
			throw new Exception();
		}
		catch
		{
		}
	}
}

Debug.WriteLine("Impossible!?!");

Thread.Sleep(Timeout.Infinite);
```

josesimoes, 2022-01-25:
Confirmed the issue. Something wrong it's happening in the execution engine that takes a wrong exit path.

josesimoes, 2022-03-08:
.NET Fiddle test app running on full .NET [here](https://dotnetfiddle.net/kslcqn).

aromaa, 2022-03-12:
What generics are you referring to? There are none? <https://sharplab.io/#v2:CYLg1APgAgTAjAWAFBQAwAIpwHQBECWAhgOYB2A9gM4Au+AxpQNzJqY4AqAFgE4CmhwfKWLMkyatwCeyAN7IAkFgBs6AEblyAG3QAxQpsq8AFAEp0AXgB86AGb7DohfhvojAQj0HjJkwrlJ5eVxeVQBXYmwAdW58al4AGSFjACIAJV5qUO5SN2STRwDFAHZReQBfZAqkGyF9TWkkf3kubnIAd1MCxTgVKAAWdBb20z8FeQkGwKbA6h529FJeNvQAUQAPOl4AB1pyUk6xqsC6Qmo6TjHpo6qq5GCwiOjYhKSjZIBJAFstqkp8VU0vDcAH5cvlkMgWvxgNgAMqA7ZGdj4T68cihajYd6kGqkZ75IA=>

Ellerbach, 2022-03-13:
> What generics are you referring to? There are none? 

Get the static function out of the finally block and it should work better, this works like a charm for me:
```csharp
using System;
using System.Threading;

static bool False() { return false; }

try
{
    if (!False())
    {
        Console.WriteLine("Return!");
        return;
    }
}
finally
{
    Throw();
}

Console.WriteLine("Impossible!?!");

Thread.Sleep(Timeout.Infinite);

static void Throw()
{
    try
    {
        throw new Exception();
    }
    catch
    {
    }
}
```

[image]

josesimoes, 2022-03-14:
@aromaa you are correct, there are no generics here. Just static functions. I was rushed on my conclusion by looking at the function names the compiler assigns when using this code style.

Ellerbach, 2023-04-18:
I'll go ahead and close this one as there is a proper way to resolve it. Feel free to reopen if you think it's not satisfying.

aromaa, 2023-04-18:
> I'll go ahead and close this one as there is a proper way to resolve it.

What does a proper way to resolve it mean here? The execution flow is broken and is literally doing **something that should be impossible**. The only way to resolve this would be to refactor your code. It took me days of headache to even narrow this down and understand what's wrong here.

aromaa, 2023-04-18:
Here is a more real world example to show the actual impact of this bug:
```csharp
using System.Diagnostics;
using System.Threading;

using CancellationTokenSource tokenSourceCancelled = new();
using CancellationTokenSource tokenSourceOther = new();

tokenSourceCancelled.Cancel();

using (tokenSourceCancelled.Token.Register(() => tokenSourceOther.Cancel()))
{
	static bool False() => false;

	if (!False())
	{
		Debug.WriteLine("Return!");

		return;
	}
}

Debug.WriteLine("Impossible!?!");

Thread.Sleep(Timeout.Infinite);
```

Ellerbach, 2023-04-18:
Please as mentionned before, taje the static function out of the try or using.

like this:

```csharp
using System.Diagnostics;
using System.Threading;

using CancellationTokenSource tokenSourceCancelled = new();
using CancellationTokenSource tokenSourceOther = new();

tokenSourceCancelled.Cancel();

static bool False() => false;

using (tokenSourceCancelled.Token.Register(() => tokenSourceOther.Cancel()))
{
	if (!False())
	{
		Debug.WriteLine("Return!");

		return;
	}
}

Debug.WriteLine("Impossible!?!");

Thread.Sleep(Timeout.Infinite);
```

Ellerbach, 2023-04-18:
> What does a proper way to resolve it mean here? The execution flow is broken and is literally doing **something that should be impossible**. The only way to resolve this would be to refactor your code. It took me days of headache to even narrow this down and understand what's wrong here.

As explained before, please take out the definition of static function from try or using o ranything like this. nanoFramework is not the full .NET, it has some compromozises that needs to be done. This is for static functions, you can define non static in functions, that should work fine.

aromaa, 2023-04-18:
I don't understand why you are referring to the use of local functions? They are just compiler syntax sugar that ends up compiling down to
```csharp
[CompilerGenerated]
internal static bool <<Main>$>g__False|0_0() => false;
```
which is equivalent of just doing it yourself by declaring it like
```csharp
internal static bool False() => false;
```

And the problem still exists even if you don't use local functions.

Ellerbach, 2023-04-18:
in nanoFramework, the static classes and functions are treated differently. Again, it's a matter of compromize. So please take out of try or using blocks the static definitions. You have the same working code as example.

aromaa, 2023-04-18:
Which code are you referring to? Even the one you linked here <https://github.com/nanoframework/Home/issues/918#issuecomment-1066113512> demonstrates the bug.

Ellerbach, 2023-04-18:
> Which code are you referring to? Even the one you linked here [#918 (comment)](https://github.com/nanoframework/Home/issues/918#issuecomment-1066113512) demonstrates the bug.

That's related to the cancellation token, not the static part. I'm referencing to the initial issue.

Ellerbach, 2023-04-18:
So taking your last example, this will work:
```csharp
using System;
using System.Diagnostics;
using System.Threading;

using CancellationTokenSource tokenSourceCancelled = new();
using CancellationTokenSource tokenSourceOther = new();

tokenSourceCancelled.Cancel();

tokenSourceCancelled.Token.Register(() =>
{
    Debug.WriteLine("Cancelled!");
    tokenSourceOther.Cancel();
});

static bool False() => false;

if (!False())
{
    Debug.WriteLine("Return!");

    return;
}
Debug.WriteLine("Impossible!?!");
Thread.Sleep(Timeout.Infinite);
```

Ellerbach, 2023-04-18:
> Yes. The bug still occurs for CancellationTokens with your [#918 (comment)](https://github.com/nanoframework/Home/issues/918#issuecomment-1513170219) suggestion here. And the initial issue reproduces too with the [#918 (comment)](https://github.com/nanoframework/Home/issues/918#issuecomment-1066113512) where you even showed a screenshot showing that it indeed is printing out the wrong result.
> 
> To make this very clear; It should never print the "Impossible!?!" to the console. The code is UNREACHABLE. It should NEVER print that. You can even see that Full Framework behaves correctly here from the Fiddler above https://dotnetfiddle.net/2VskfP

Ho, I get it now. Sorry. OK, so reopening this issue.

aromaa, 2023-04-18:
No worries. The bug is a very niche edge case with some odd looking code to repro it :)

closed 2026-10-01 by josesimoes