nanoframework/Home #923: ESP32 nightly build failing

networkfusion, 2021-12-31:
### Details about Problem
[image]

**Target:**
All ESP32 nightly builds

**Firmware image version:**
Latest.

Worked before? If so, with which nanoFramework image version:

**Device capabilities output:**

### Description

Relivent bit is:

```extern_ram_seg:          0 GB         4 MB      0.00c:/users/vssadministrator/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: nanoCLR.elf section `.iram0.text' will not fit in region `iram0_0_seg'
c:/users/vssadministrator/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: IRAM0 segment data does not fit.
c:/users/vssadministrator/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `iram0_0_seg' overflowed by 1156 bytes
collect2.exe: error: ld returned 1 exit status
```

### Detailed repro steps so we can see the same problem

1.

2.

...

### Other suggested things
<!-- if applicable/relevant -->

### Expected behaviour

A clear and concise description of what you expected to happen.

### Screenshot
<!-- if applicable/relevant -->
<!--Very helpful if you send along a few screenshots to help visualize the issue!-->

### Additional context

Add any other context about the problem here.

### Make an effort to fix the bug

Attempt to submit a [Pull Request (PR)](https://help.github.com/articles/about-pull-requests/) that fixes the bug. Include in this PR a test that verifies the fix. If you were not able to fix the bug, a PR that illustrates your partial progress will suffice.

<!-- bug-report-fw-tag DO NOT REMOVE -->

networkfusion, 2021-12-31:
Perhaps related with https://github.com/nanoframework/Home/issues/919

josesimoes, 2022-01-03:
Reported on Discord Core Team channel after the move to IDF 4.3.2. Request @AdrianSoundy to look at this.

closed 2022-01-04 by josesimoes