nanoframework/Home #925: Installing nanoFramework.M5Core2 or nanoFramework.M5Core preview nuget packages causes assembly resolve errors

mikeoliphant, 2022-01-01:
### Details about Problem

Installing nanoFramework.M5Core2 or nanoFramework.M5Core preview nuget packages causes applications to fail to run with **"Link failure: some assembly references cannot be resolved!!"**

**.NET nanoFramework library:**  nanoFramework.M5Core2 and nanoFramework.M5Core 

**VS version:** VS 2022 Version 17.0.4

**VS extension version:** nanoFramework Extensions v2022.1.0.6

### Description

Looks like maybe an assembly version mismatch in the nuget deps? Relevant bit from the debug log:

```
Resolving.
Link failure: some assembly references cannot be resolved!!

Assembly: nanoFramework.M5Core2 (0.0.0.0) needs assembly 'Iot.Device.Rtc' (1.0.0.0)
Assembly: nanoFramework.M5Core2 (0.0.0.0) needs assembly 'Iot.Device.Axp192' (1.0.0.0)
Assembly: nanoFramework.M5Core2 (0.0.0.0) needs assembly 'Iot.Device.Mpu6886' (1.0.0.0)
Assembly: Iot.Device.Mpu6886 (1.0.0.0) needs assembly 'UnitsNet.Temperature' (4.110.0.0)
Assembly: Iot.Device.Ip5306 (0.0.0.0) needs assembly 'UnitsNet.ElectricCurrent' (4.110.0.0)
Assembly: Iot.Device.Axp192 (1.0.0.0) needs assembly 'UnitsNet.ElectricPotential' (4.110.0.0)
Assembly: Iot.Device.Axp192 (1.0.0.0) needs assembly 'UnitsNet.ElectricCurrent' (4.110.0.0)
Assembly: Iot.Device.Axp192 (1.0.0.0) needs assembly 'UnitsNet.Temperature' (4.110.0.0)
Assembly: Iot.Device.Axp192 (1.0.0.0) needs assembly 'UnitsNet.Power' (4.110.0.0)
Assembly: Iot.Device.Rtc (1.0.0.0) needs assembly 'UnitsNet.Temperature' (4.110.0.0)
Error: a3000000
```

I am running on an M5Stack Core2 with nanoframework installed by:

`nanoff --target M5Core2 --update --preview --serialport COM3`

### Detailed repro steps so we can see the same problem

1. Create new blank ("Hello") nanoframework project.

2. Verify that it deploys and runs properly

3. Install pre-release version of nanoFramework.M5Core2 or nanoFramework.M5Core nuget packages

4. Verify that the application fails to run with assembly resolution errors.

mikeoliphant, 2022-01-02:
It turns out that nanoFramework.M5Core2 v1.0.0 works, so maybe the preview nuget packages aren't aligned with the preview firmware?

sukesh-ak, 2022-01-03:
> It turns out that nanoFramework.M5Core2 v1.0.0 works, so maybe the preview nuget packages aren't aligned with the preview firmware?

In the case of M5Stack to M5Core, there is also a rename portion. 
So not sure if its just the preview.

cristian-recoseanu, 2022-01-03:
I am also currently experiencing this on ESP32 preview when using UnitsNet.Temperature through the AHTXX package.

mikeoliphant, 2022-01-07:
I think this can be fixed by updating the various Iot.Device.XXX nuget packages to require v4.112.0 of the relevant UnitsNet packages. Not sure why it seems to have to exactly match the version, when they are requesting version >= 4.110.0, though...

For now, you can work around the problem by manually downgrading the UnitsNet nuget packages for v4.110.0.

josesimoes, 2022-01-08:
That's the workaround for the moment.
The references update tool has been fixed in the meantime and this will be addresses shortly.

josesimoes, 2022-01-11:
Fixed with nanoframework/nanoFramework.M5Stack#51.

closed 2022-01-11 by josesimoes

mikeoliphant, 2022-01-11:
Working now. Thanks @josesimoes!