nanoframework/Home #927: ESP32:ESPNOW feature

SandorDobos, 2022-01-03:
**nanoFramework area:** Community contributions

### Is your feature request related to a problem? Please describe.
ESP32 has a radio communication method (ESPNOW) which allows peer-to-peer, peer-to-many, 
many-to-peer and many-to-many data exchange without any infrastructure (e.g. no access point needed).
This allows many uses between equal level nodes.

### Describe the solution you'd like
Lets have ESPNOW available in managed code.

### Describe alternatives you've considered
Wifi and BLE. Both lacking some of features mentioned above.

### Additional context
[The official ESPNOW documentation is here.](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_now.html?highlight=espnow)

### Submit a PR with the implementation
My implementation lives here.
[Native part](https://github.com/SandorDobos/nf-interpreter/tree/feature-nanoFramework.Hardware.Esp32.EspNow)
[Managed code](https://github.com/SandorDobos/nanoFramework.Hardware.Esp32.EspNow)
I am sure some administration is still missing, but technically the feature is fully implemented.
Please point me to necessary steps required to accept this implementation!

AdrianSoundy, 2022-01-05:
This looks like a good addition to the nanoFramework. Your project looks mainly ok except for the Initialization of the WiFi. As it possible to run ESP-NOW and a Wifi station or AP the WiFi init will need to be left up to nanoFramework. Maybe have a check that the WiFI is in correct state before dong the esp_now_init, but Init & deinit needs to be done by system and esp_now_init() & esp_now_deinit can be done by the naitive part of the esp_now.

We will need to create a repo for the nanoFramework.Hardware.Esp32.EspNow so you can clone and update with your managed code changes,

For the native part clone current nanoFramework, add your changes and then create a PR for changes.

AdrianSoundy, 2022-01-05:
I have created a nanoFramework.Hardware.Esp32.EspNow repo which you can now clone, update and create PR.
Use another repo as template for required files etc.

AdrianSoundy, 2022-01-05:
nanoFramework.Hardware.Esp32.EspNow is probably not the best choice for this assembly as it not specific to any hardware component in ESP32. Will rename repo once name is decided on.

SandorDobos, 2022-01-06:
> This looks like a good addition to the nanoFramework. Your project looks mainly ok except for the Initialization of the WiFi. As it possible to run ESP-NOW and a Wifi station or AP the WiFi init [...]

I would remove wifi init from native code and add "wifi initialized somehow" prerequisite to the managed package.
But I found no ways to just init wifi (equivalent of code to be removed from my native implementation).
I found just samples to pull up full fladged AP or STA.
What if the developer do not want AP nor STA just ESPNOW?
Any suggestions appreciated :)

SandorDobos, 2022-01-06:
Wifi init code removed. It turns out that steps start from ESP's nanoHAL_Initialize already do enough for ESPNOW to work, so no additional code required :)

AdrianSoundy, 2022-01-07:
It has been decided to use the nanoFramework.EspNow namespace for this feature so repo has been renamed
@SandorDobos

SandorDobos, 2022-01-17:
@AdrianSoundy  native and managed code reflects the namespace change now

SandorDobos, 2022-04-16:
Not stale just waiting for @AdrianSoundy

AdrianSoundy, 2022-04-17:
I don't see any PR for the hardware.Esp32now repo
@SandorDobos

TheMo, 2024-04-16:
Hi,
What is the status of Esp32now in nanoframework? Could you please give us an update?

josesimoes, 2026-09-29:
Please note that this has just been added.