nanoframework/Home #935: DeviceClient never confirms message delivery

aromaa, 2022-01-09:
### Details about Problem

**.NET nanoFramework library:** nanoFramework.Azure.Devices / nanoFramework.m2mqtt

### Description
DeviceClient never confirms delivery of SendMessage or UpdateReportedProperties and waits for the cancellation token cancellation.

Ellerbach, 2022-01-10:
Can you please give the code you are using?
If you are using AtMostOnce (0), it's normal you'll never ever receive any confirmation as in this mode, you'll not get any!
So you need level 1 to get a confirmation.

aromaa, 2022-01-10:
Added the repro code, but yes, this doesn't work even when using AtLeastOnce. And UpdateReportedProperties uses always AtLeastOnce so the qos level only applies to SendMessage.

Ellerbach, 2022-01-10:
> ```cs
> AtLeastOnce
> ```

Let me try again in next days. The only issue I saw in the past was when the network is too fast. You get the confirmation before it is added in the queue. All is using the same principal. So that *maybe* your case.

Ellerbach, 2022-01-10:
> Added the repro code, but yes, this doesn't work even when using AtLeastOnce. And UpdateReportedProperties uses always AtLeastOnce so the qos level only applies to SendMessage.

Also, just to be 100% sure, you're using the latest preview version?

aromaa, 2022-01-10:
Yes, using the latest preview version. The network conditions are indeed good and suspected it too when I took a quick glance on the code.

Ellerbach, 2022-01-10:
> Yes, using the latest preview version. The network conditions are indeed good and suspected it too when I took a quick glance on the code.

So seems it's the network which is too fast :-). I will look at how to kind of force to get it and adjust the code to add it before. Will play with that in next few days.

Ellerbach, 2022-01-11:
@aromaa can you please check if the PR https://github.com/nanoframework/nanoFramework.Azure.Devices/pull/65 fixes your issue? Clone the repo, checkout the branch and rather than using the nuget, if you can use that code, would be great.
I'll test it as well later.

aromaa, 2022-01-11:
Seems not to be enough, not getting any confirmations though

Ellerbach, 2022-01-14:
> Seems not to be enough, not getting any confirmations though

Good news: I manage to reproduce the issue. Bad news: that's insane to debug!

Ellerbach, 2022-01-14:
@aromaa feel free to test the new version, all should now work fine. I've tested your example, works like a charm for me again in all circumstances. Thanks for finding out and thanks for the sample code!

aromaa, 2022-01-14:
Can confirm that the issue has been fixed! Yet another thing off backlog :)