nanoframework/Home #940: Modify device hostname

hyper-wave, 2022-01-12:
:exclamation: Remove the content above here and fill out details below. :exclamation:

**nanoFramework area:** (Visual Studio extension | Other tools | Hardware/target board | Community targets | Community contributions | Documentation )

### Is your feature request related to a problem? Please describe.
I am unable to change the hostname, so multiple devices on the same network show the same hostname. 

### Describe the solution you'd like
we would like the ability to change the device hostname. 

### Describe alternatives you've considered
there are no alternatives that we are aware of. 

### Additional context
Add any other context or screenshots about the feature request here.

### Submit a PR with the implementation

If you know how to add whatever you're suggesting, let us know and have a [Pull Request (PR)](https://help.github.com/articles/about-pull-requests/) ready. 
If you are uncertain if the feature will be accepted, outline the proposal to confirm it is viable before commiting time and effort to it.

Even if the feature is a good idea and viable, it may not be accepted since the ongoing effort in maintaining the feature may outweigh the benefit it delivers.

josesimoes, 2022-01-12:
@hyper-wave I've fixed the issue for you 😜

aromaa, 2022-01-13:
Something nice to have would be the ability to use the device certificate subject as the hostname. And then getter & setter for this so it can be known to check what the device is named.

josesimoes, 2022-01-24:
> Something nice to have would be the ability to use the device certificate subject as the hostname. And then getter & setter for this so it can be known to check what the device is named.

Agreed that we should expose an interface to get&set the host name.

closed 2022-01-24 by josesimoes

DarkNoir, 2024-01-02:
So, there is now way to set the host name as of Jan 2024?

Ellerbach, 2024-01-03:
> So, there is now way to set the host name as of Jan 2024?

No, the name is composed by `nanodevice_ABCDEF` where ABCDEF are the 6 first digit of the MAC address