nanoframework/Home #942: Have an ability to change the Manufacturer Name and Model of Bluetooth Low Energy.

alberk8, 2022-01-13:
**nanoFramework area:** (Visual Studio extension | Other tools | Hardware/target board | Community targets | Community contributions | Documentation )

### Is your feature request related to a problem? Please describe.
The default Manufacturer Name and Model is set in Native Code.

### Describe the solution you'd like
Have the ability to change the Bluetooth Low Energy (BLE) Manufacturer Name and Model Number in C#.

### Describe alternatives you've considered

### Additional context
The Native code is set in this file [https://github.com/nanoframework/nf-interpreter/blob/develop/targets/ESP32/_nanoCLR/nanoFramework.Device.Bluetooth/sys_ble_nimble_chars.c](url)
```
static const char *manuf_name = "Nanoframework";
static const char *model_num = "ESP32";
```

### Submit a PR with the implementation

hyper-wave, 2022-01-25:
It would be nice to have to ability to change not only those values, but leverage the rest of the Device Information UUID's 

2A25 - Serial Number
2A26 - Firmware Revision
2A27 - Hardware Revision
2A28 - Software Revision

AdrianSoundy, 2022-02-25:
The windows uwp Windows.Device.Bluetooth doesn't have a way of changing the Device Information which means we will need to expand it to allow acess to it so these can be changed or new UUID's added.

AdrianSoundy, 2022-03-01:
The Device Information is a separate service which is added by native code when main service is created.
I can't see a way in the UWP WIndows.Device.Bluetooth to define more than 1 service from managed code.

# I can either pass the device information from managed code and continue to do this is native code. 
# Create the extra services attached to primary service in managed code and just create them in native code (more flexible) and an are that seems to be lacking in UWP version. (TODO)

Todo
Add a Services property to GattLocalServiceProvider. When Primary service created it will automatically create the Device information service with default values. These values can be updated and extra characteristics added to service.

AdrianSoundy, 2022-03-06:
The UWP version doesn't allow user to add extra services to Profile (GattLocalServiceProvider). You just have 1 service and the Devcie Information added automatically.

Made some changes to allow servives to be added GattLocalServiceProvider.
When you create a new GattLocalServiceProvider it will now create the primary service and Device Information Service in managed code.

The following new methods have been added to GattLocalServiceProvider:-
- Services() - Return array of services associated with GattLocalServiceProvider
- AddService(uuid) - Add a new service to GattLocalServiceProvider, must not already exist.
- FindService(UUID) - Returns GattLocalService for Uuid or null

This will allow the Device Information to be changed and to add other services like "Battery Service"

AdrianSoundy, 2022-03-13:
Because not so easy to modify a service after its created without having lots of setters on classes decided to have AddService replace an exiting service if it exists.

This means you can just create a new service and add it to provider.

For this issue you create your own Device Information Service, add it and it will replace default Device Information. 
Have dropped FindService() method as no longer needed.

Have created a few sample Service classes used in testing which will be put into sample which will be useful to everybody.
- DeviceInformation Service
- Battery Service ( expose level of battery )
- CurrentTime Service (Expose current time on device and set date time)
- Environmental Sensor Service

closed 2022-03-16 by AdrianSoundy