nanoframework/Home #951: OLIMEX has no support for ethernet adapter

espijk2, 2022-01-20:
Although not mentioned in the firmware [matrix ](https://docs.nanoframework.net/content/esp32/esp32_firmware_versions.html) the [OLIMEX Gateway board](https://www.olimex.com/Products/IoT/ESP32/ESP32-GATEWAY/), with a ethernet adapter should work accroding to @josesimoes when targetting ESP32_OLIMEX

NetworkInterface.GetAllNetworkInterfaces(); returns only return the Wireless80211 and WirelessAP.
The leds of switch port and adapter are not showing any activity either.

Issue might be related to recent move to IDF 4.3

espijk2, 2022-01-21:
With an [OLIMEX EVB](https://www.olimex.com/Products/IoT/ESP32/ESP32-EVB/) board have the same results, Although network leds are on!
NetworkInterface.GetAllNetworkInterfaces() results only in WiFiWireless and Wireless AP no Adapter and not able to connect with wireless.

OniVe, 2022-01-25:
> NetworkInterface.GetAllNetworkInterfaces(); returns only return the Wireless80211 and WirelessAP.
The leds of switch port and adapter are not showing any activity either.

The [LilyGO T-ETH-POE](http://www.lilygo.cn/prod_view.aspx?TypeId=50063&Id=1307&FId=t3:50063:3) board shows the same behavior.

AdrianSoundy, 2022-01-31:
In the latest builds (IDF 4.3.x) the Ethernet driver is not being included for all builds.
This will be corrected shortly

closed 2022-02-08 by AdrianSoundy

CZEMacLeod, 2022-05-19:
> Not able to create any network connectivity also no Wireless. Network Helper for wifi results in no WiFi connection and the wifiAdapter.ScanAsync(); results in an internal Exception.

Using the latest firmware and packages, ethernet seems to work on the OLIMEX ESP32-POE-ISO board I have, but I get the same issue with `wifiAdapter.ScanAsync();` resulting in an Exception (while no ethernet cable attached).
```
    ++++ Exception System.InvalidOperationException - CLR_E_INVALID_OPERATION (1) ++++
    ++++ Message: 
    ++++ System.Device.Wifi.WifiAdapter::NativeScanAsync [IP: 0000] ++++
    ++++ System.Device.Wifi.WifiAdapter::ScanAsync [IP: 000d] ++++
```

josesimoes, 2022-05-19:
@CZEMacLeod check [this thread](https://discord.com/channels/478725473862549535/974640049704538213/976920292876222564) at our Discord server. I believe that will help you.