nanoframework/Home #955: Get Enum member as Named string

alberk8, 2022-01-24:
**nanoFramework area:** (Visual Studio extension | Other tools | Hardware/target board | Community targets | Community contributions | Documentation )

### Is your feature request related to a problem? Please describe.
Return the member of  Enum Name as a string.  Current the returned string is the string representation of the value.

### Describe the solution you'd like
`Console.WriteLine(Color.Red.ToString());`

[Example](https://dotnetfiddle.net/bwNG6Ql)

### Describe alternatives you've considered

### Additional context

### Submit a PR with the implementation

If you know how to add whatever you're suggesting, let us know and have a [Pull Request (PR)](https://help.github.com/articles/about-pull-requests/) ready. 
If you are uncertain if the feature will be accepted, outline the proposal to confirm it is viable before commiting time and effort to it.

Even if the feature is a good idea and viable, it may not be accepted since the ongoing effort in maintaining the feature may outweigh the benefit it delivers.

Ellerbach, 2022-01-24:
Hi @alberk8 I'm afraid this will not be possible. See here the explanations: https://docs.nanoframework.net/content/architecture/simplifications-and-trade-offs.html#enums

In case this would be added, the size of an enum will drastically increase and use a lot of space both in the flash and in memory. All the use case can easily be done thru a simple function returning a string when it's needed.

If you need those, then just add a function in your code to return them.
so closing this issue.

closed 2022-01-24 by Ellerbach

DarkNoir, 2024-02-14:
I simply added an extension method for the few enums I wanted to do this with. So far, I am happy with the compromises made to keep the framework compact. More room for my code :).