nanoframework/Home #959: Environment.TickCount64 breaks nanoCLR WIN32 (TestFramework 1.0.173)

mboud, 2022-01-30:
### Details about Problem

**.NET nanoFramework library:**  nanoFramework.TestFramework 1.0.173

**VS version:** 2019

### Description
It seems `Environment.TickCount64` usage breaks the unit tests. For example, simply accessing it and putting its value into a variable will break my tests. Despite willing to run all tests, it will stop after failing the first one, without usable trace info.

### Detailed repro steps so we can see the same problem
Create one or more tests using `Environment.TickCount64` directly or through an method from a custom library. Try and run tests, the first one will fail and the other ones won't run.

### Expected behaviour
Main issue: `Environment.TickCount64` should not break tests.
Bonus: Tests should all run despite the error.

### Additional context
Event viewer details:
```
Faulting application name: nanoFramework.nanoCLR.exe, version: 0.0.0.0, time stamp: 0x61f3d3a6
Faulting module name: nanoFramework.nanoCLR.exe, version: 0.0.0.0, time stamp: 0x61f3d3a6
Exception code: 0xc0000005
Fault offset: 0x00034b3e
Faulting process id: 0x21d4
Faulting application start time: 0x01d815f6fa4ac1d0
Faulting application path: D:\arduino\ESP32\TinyGPSPlusNF\src\packages\nanoFramework.TestFramework.1.0.173\lib\net48\nanoFramework.nanoCLR.exe
Faulting module path: D:\arduino\ESP32\TinyGPSPlusNF\src\packages\nanoFramework.TestFramework.1.0.173\lib\net48\nanoFramework.nanoCLR.exe
Report Id: 8bd9e870-988a-42e8-9ad4-9e3d0c3f0f3c
Faulting package full name: 
Faulting package-relative application ID: 
```

Failed test output:
```
   .NET nanoFramework nanoCLR WIN32 v1.7.3-preview.89+b3f568f524

Copyright (c) 2020 .NET Foundation and Contributors

Starting...
Created EE.
Started Hardware.
Loading Assemblies.
   Assembly: NFUnitTest (1.0.0.0)   (692 RAM - 10628 ROM - 2261 METADATA)

   AssemblyRef    =       12 bytes (       3 elements)
   TypeRef        =       96 bytes (      24 elements)
   FieldRef       =       40 bytes (      10 elements)
   MethodRef      =      204 bytes (      51 elements)
   TypeDef        =       48 bytes (       6 elements)
   FieldDef       =        4 bytes (       1 elements)
   MethodDef      =       96 bytes (      48 elements)
   StaticFields   =        0 bytes (       0 elements)

   Attributes      =      352 bytes (      44 elements)
   TypeSpec        =        0 bytes (       0 elements)
   Resources       =        0 bytes (       0 elements)
   Resources Files =        0 bytes (       0 elements)
   Resources Data  =        0 bytes
   Strings         =     4612 bytes
   Signatures      =      273 bytes
   ByteCode        =     3627 bytes

   Assembly: System.Math (1.4.4.0)   (300 RAM - 1104 ROM - 750 METADATA)

   AssemblyRef    =        4 bytes (       1 elements)
   TypeRef        =       16 bytes (       4 elements)
   FieldRef       =        0 bytes (       0 elements)
   MethodRef      =       20 bytes (       5 elements)
   TypeDef        =        8 bytes (       1 elements)
   FieldDef       =        0 bytes (       0 elements)
   MethodDef      =       72 bytes (      36 elements)
   StaticFields   =        0 bytes (       0 elements)

   Attributes      =        0 bytes (       0 elements)
   TypeSpec        =        0 bytes (       0 elements)
   Resources       =        0 bytes (       0 elements)
   Resources Files =        0 bytes (       0 elements)
   Resources Data  =        0 bytes
   Strings         =       37 bytes
   Signatures      =       66 bytes
   ByteCode        =      188 bytes

   Assembly: TinyGPSPlusNF (1.0.0.0)   (820 RAM - 7828 ROM - 2936 METADATA)

   AssemblyRef    =       12 bytes (       3 elements)
   TypeRef        =       52 bytes (      13 elements)
   FieldRef       =        0 bytes (       0 elements)
   MethodRef      =       92 bytes (      23 elements)
   TypeDef        =      120 bytes (      15 elements)
   FieldDef       =      104 bytes (      51 elements)
   MethodDef      =      196 bytes (      98 elements)
   StaticFields   =        0 bytes (       0 elements)

   Attributes      =        0 bytes (       0 elements)
   TypeSpec        =        0 bytes (       0 elements)
   Resources       =        0 bytes (       0 elements)
   Resources Files =        0 bytes (       0 elements)
   Resources Data  =        0 bytes
   Strings         =     1751 bytes
   Signatures      =      276 bytes
   ByteCode        =     3013 bytes

   Assembly: mscorlib (1.12.0.0)   (4100 RAM - 32396 ROM - 19575 METADATA)

   AssemblyRef    =        0 bytes (       0 elements)
   TypeRef        =        0 bytes (       0 elements)
   FieldRef       =        0 bytes (       0 elements)
   MethodRef      =        0 bytes (       0 elements)
   TypeDef        =     1160 bytes (     145 elements)
   FieldDef       =      196 bytes (      98 elements)
   MethodDef      =     1636 bytes (     817 elements)
   StaticFields   =      144 bytes (      12 elements)

   Attributes      =       40 bytes (       5 elements)
   TypeSpec        =        4 bytes (       1 elements)
   Resources       =        0 bytes (       0 elements)
   Resources Files =        0 bytes (       0 elements)
   Resources Data  =        0 bytes
   Strings         =     2716 bytes
   Signatures      =     2195 bytes
   ByteCode        =     9979 bytes

   Assembly: nanoFramework.Runtime.Native (1.5.4.0)   (408 RAM - 1568 ROM - 970 METADATA)

   AssemblyRef    =        4 bytes (       1 elements)
   TypeRef        =       40 bytes (      10 elements)
   FieldRef       =        0 bytes (       0 elements)
   MethodRef      =       48 bytes (      12 elements)
   TypeDef        =       72 bytes (       9 elements)
   FieldDef       =        4 bytes (       2 elements)
   MethodDef      =       56 bytes (      27 elements)
   StaticFields   =       12 bytes (       1 elements)

   Attributes      =        0 bytes (       0 elements)
   TypeSpec        =        0 bytes (       0 elements)
   Resources       =        0 bytes (       0 elements)
   Resources Files =        0 bytes (       0 elements)
   Resources Data  =        0 bytes
   Strings         =      306 bytes
   Signatures      =      118 bytes
   ByteCode        =      163 bytes

   Assembly: nanoFramework.TestFramework (1.0.173.0)   (632 RAM - 5520 ROM - 1898 METADATA)

   AssemblyRef    =        4 bytes (       1 elements)
   TypeRef        =       96 bytes (      24 elements)
   FieldRef       =        0 bytes (       0 elements)
   MethodRef      =      112 bytes (      28 elements)
   TypeDef        =       80 bytes (      10 elements)
   FieldDef       =        4 bytes (       1 elements)
   MethodDef      =      120 bytes (      60 elements)
   StaticFields   =       12 bytes (       1 elements)

   Attributes      =       16 bytes (       2 elements)
   TypeSpec        =        0 bytes (       0 elements)
   Resources       =        0 bytes (       0 elements)
   Resources Files =        0 bytes (       0 elements)
   Resources Data  =        0 bytes
   Strings         =     1148 bytes
   Signatures      =      246 bytes
   ByteCode        =     2346 bytes

   Assembly: nanoFramework.UnitTestLauncher (0.0.0.0)   (304 RAM - 1288 ROM - 481 METADATA)

   AssemblyRef    =        8 bytes (       2 elements)
   TypeRef        =       68 bytes (      17 elements)
   FieldRef       =        0 bytes (       0 elements)
   MethodRef      =       64 bytes (      16 elements)
   TypeDef        =        8 bytes (       1 elements)
   FieldDef       =        0 bytes (       0 elements)
   MethodDef      =        8 bytes (       3 elements)
   StaticFields   =        0 bytes (       0 elements)

   Attributes      =        0 bytes (       0 elements)
   TypeSpec        =        0 bytes (       0 elements)
   Resources       =        0 bytes (       0 elements)
   Resources Files =        0 bytes (       0 elements)
   Resources Data  =        0 bytes
   Strings         =      282 bytes
   Signatures      =      121 bytes
   ByteCode        =      395 bytes

Resolving.

Total: (6160 RAM - 60332 ROM - 28871 METADATA)

   AssemblyRef    =       44 bytes (      11 elements)
   TypeRef        =      368 bytes (      92 elements)
   FieldRef       =       40 bytes (      10 elements)
   MethodRef      =      540 bytes (     135 elements)
   TypeDef        =     1496 bytes (     187 elements)
   FieldDef       =      312 bytes (     153 elements)
   MethodDef      =     2184 bytes (    1089 elements)
   StaticFields   =      168 bytes (      14 elements)

   DebuggingInfo  =     1096 bytes

   Attributes      =      408 bytes (      51 elements)
   TypeSpec        =        4 bytes (       1 elements)
   Resources Files =        0 bytes (       0 elements)
   Resources       =        0 bytes (       0 elements)
   Resources Data  =        0 bytes
   Strings         =    10852 bytes
   Signatures      =     3295 bytes
   ByteCode        =    19711 bytes

Ready.
```

test-in-prod, 2022-01-31:
Maybe similar, but accessing `Environment.TickCount64` on an ESP32 chip (ESP32-WROOM32) will crash it.

jonstelly, 2022-01-31:
FWIW, TickCount64 doesn't crash/throw when debugging on my ESP32-WROOM-32U ([this one](https://www.amazon.com/gp/product/B09BM1QW29)).  

Code inside Main (`Log.WriteLine` just writes to Console with a timestamp):

    Log.WriteLine($"Tick Count: {Environment.TickCount64}");

System Info:

```System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF v4.3.2
  Target:   ESP32
  Platform: ESP32

Firmware build Info:
  Date:        Jan 29 2022
  Type:        MinSizeRel build, chip rev. >= 0, without support for PSRAM
  CLR Version: 1.7.3.252
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

Assemblies:
  Pool.Controller, 1.0.0.0
  Windows.Storage.Streams, 1.12.4.0
  System.Net, 1.8.0.0
  mscorlib, 1.12.0.0
  System.Device.Gpio, 1.0.3.0

Native Assemblies:
  mscorlib v100.5.0.16, checksum 0xE7505F22
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.7.3, checksum 0xBE7FF253
  nanoFramework.Hardware.Esp32.Rmt v100.0.3.0, checksum 0x9A53BB44
  nanoFramework.Devices.OneWire v100.0.3.4, checksum 0xA5C172BD
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.0.1, checksum 0x5A31313D
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x210D05B1
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.4.0, checksum 0x5B533478
  Windows.Devices.Adc v100.1.3.3, checksum 0xCA03579A
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.4, checksum 0xB6D0ACC1
  Windows.Devices.Gpio v100.1.2.2, checksum 0xC41539BE
  Windows.Devices.I2c v100.2.0.2, checksum 0x79EDBF71
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  Windows.Devices.Pwm v100.1.3.3, checksum 0xBA2E2251
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  Windows.Devices.SerialCommunication v100.1.1.2, checksum 0x34BAF06E
  System.IO.Ports v100.1.2.0, checksum 0x564F2452
  Windows.Devices.Spi v100.1.4.2, checksum 0x360239F1
  System.Device.Spi v100.1.0.0, checksum 0x48031DC5
  System.Device.WiFi v100.0.6.3, checksum 0x5C9E06C4
  Windows.Storage v100.0.2.0, checksum 0x954A4192
```

test-in-prod, 2022-02-01:
Thanks @josesimoes @jonstelly I'll try to collect some more data next time I get around to it.

closed 2022-02-02 by josesimoes

test-in-prod, 2022-02-06:
Just had a chance to collect some data. I have the following as the first line to be executed, I can set a breakpoint on that line and it will pause when debugging.

```csharp
Debug.WriteLine($"tick count = {Environment.TickCount64}");
```

Stepping into/over it will cause debugging to crash with the following written to output.

```
Resolving.

Total: (13120 RAM - 115300 ROM - 56629 METADATA)

   AssemblyRef    =      124 bytes (      31 elements)
   TypeRef        =     1144 bytes (     286 elements)
   FieldRef       =       24 bytes (       6 elements)
   MethodRef      =     1492 bytes (     373 elements)
   TypeDef        =     2688 bytes (     336 elements)
   FieldDef       =      944 bytes (     466 elements)
   MethodDef      =     3916 bytes (    1952 elements)
   StaticFields   =      828 bytes (      69 elements)

   DebuggingInfo  =     1976 bytes

   Attributes      =      128 bytes (      16 elements)
   TypeSpec        =        8 bytes (       2 elements)
   Resources Files =        0 bytes (       0 elements)
   Resources       =        0 bytes (       0 elements)
   Resources Data  =        0 bytes
   Strings         =    21784 bytes
   Signatures      =     7777 bytes
   ByteCode        =    35097 bytes

Ready.
```

test-in-prod, 2022-02-06:
Just had a chance to collect some data. I have the following as the first line to be executed, I can set a breakpoint on that line and it will pause when debugging.

```csharp
Debug.WriteLine($"tick count = {Environment.TickCount64}");
```

Stepping into/over it will cause debugging to crash with the following written to output.

```
Resolving.

Total: (13120 RAM - 115300 ROM - 56629 METADATA)

   AssemblyRef    =      124 bytes (      31 elements)
   TypeRef        =     1144 bytes (     286 elements)
   FieldRef       =       24 bytes (       6 elements)
   MethodRef      =     1492 bytes (     373 elements)
   TypeDef        =     2688 bytes (     336 elements)
   FieldDef       =      944 bytes (     466 elements)
   MethodDef      =     3916 bytes (    1952 elements)
   StaticFields   =      828 bytes (      69 elements)

   DebuggingInfo  =     1976 bytes

   Attributes      =      128 bytes (      16 elements)
   TypeSpec        =        8 bytes (       2 elements)
   Resources Files =        0 bytes (       0 elements)
   Resources       =        0 bytes (       0 elements)
   Resources Data  =        0 bytes
   Strings         =    21784 bytes
   Signatures      =     7777 bytes
   ByteCode        =    35097 bytes

Ready.
```

-------

```
System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF v4.3.2
  Target:   ESP32
  Platform: ESP32

Firmware build Info:
  Date:        Jan 29 2022
  Type:        MinSizeRel build, chip rev. >= 0, without support for PSRAM
  CLR Version: 1.7.3.252
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  test, 1.0.0.0
  System.Diagnostics.Stopwatch, 1.0.0.0
  System.Net, 1.8.0.0
  nanoFramework.Runtime.Native, 1.5.4.0
  nanoFramework.Hardware.Esp32, 1.3.5.0
  System.Device.Gpio, 1.0.3.0
  nanoFramework.Runtime.Events, 1.10.0.0
  System.IO.Ports, 1.0.3.0
  System.Math, 1.4.4.0
  System.Threading, 1.0.4.0
  Windows.Storage.Streams, 1.12.4.0
  mscorlib, 1.12.0.0
  Windows.Storage, 1.4.6.0
  nanoFramework.System.Text, 1.1.3.0

Native Assemblies:
  mscorlib v100.5.0.16, checksum 0xE7505F22
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.7.3, checksum 0xBE7FF253
  nanoFramework.Hardware.Esp32.Rmt v100.0.3.0, checksum 0x9A53BB44
  nanoFramework.Devices.OneWire v100.0.3.4, checksum 0xA5C172BD
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.0.1, checksum 0x5A31313D
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x210D05B1
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.4.0, checksum 0x5B533478
  Windows.Devices.Adc v100.1.3.3, checksum 0xCA03579A
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.4, checksum 0xB6D0ACC1
  Windows.Devices.Gpio v100.1.2.2, checksum 0xC41539BE
  Windows.Devices.I2c v100.2.0.2, checksum 0x79EDBF71
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  Windows.Devices.Pwm v100.1.3.3, checksum 0xBA2E2251
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  Windows.Devices.SerialCommunication v100.1.1.2, checksum 0x34BAF06E
  System.IO.Ports v100.1.2.0, checksum 0x564F2452
  Windows.Devices.Spi v100.1.4.2, checksum 0x360239F1
  System.Device.Spi v100.1.0.0, checksum 0x48031DC5
  System.Device.WiFi v100.0.6.3, checksum 0x5C9E06C4
  Windows.Storage v100.0.2.0, checksum 0x954A4192
```

test-in-prod, 2022-02-06:
Ok, verified that the most recent firmware version 1.7.4-preview.1 is no longer crashing :-)