nanoframework/Home #960: Documentation should be generated against `stable` as well as (the current) `preview` branch

networkfusion, 2022-02-02:
### Details about Problem
Currently all documentation is generated against `preview` libs and firmware (`develop` branch),
We should also generate documentation against `stable` releases, and then VERSION it, so it can be referred to later.

**nanoFramework area:** 
Documentation (github-pages)

**VS version**
N/A

**VS extension version:**
N/A

**Target:**
N/A

**Firmware image version:**
N/A

### Description

josesimoes, 2022-02-03:
I don't know how to accomplish this without an immense amount of work on top of docfx...

At this time we're making small incremental changes on existing libraries so there aren't that big differences between stable and preview.

Honestly: don't think it's worth the effort.

networkfusion, 2022-02-03:
Understood, but think we should leave this as a future improvement (feature request). TBH, it should default to `stable` in most situations IMHO, but we can discuss that in discord at a later date.

Ellerbach, 2022-02-04:
I have to agree with @josesimoes that it's not really worth the effort. The tags on DocFX configuration won't really help as it's the pipeline which defines which branch is build. In our case, always the develop one.
But for sure, we can improve what is written on this page https://docs.nanoframework.net/api/index.html to explain it's always the develop one which is built. That the build does not happen real time neither and that there may be very small differences.

closed 2022-02-04 by networkfusion