nanoframework/Home #961: OneWireController() does not take pins change into account and is hard coded with pins 16 and 17

jefmad, 2022-02-03:
### Details about Problem

**.NET nanoFramework library:** nanoFramework.Devices.OneWire

**VS version:**  2019 

**VS extension version:**

**Target:** Tested on several models of ESP32

**Device capabilities output:**

### Description
OneWireController() is hard coded with pin 16 and pin 17 (RX2 and TX2) and the following call before is not taken into account.
            Configuration.SetPinFunction(5, DeviceFunction.COM2_RX);
            Configuration.SetPinFunction(18, DeviceFunction.COM2_TX);

This is a problem when using ESP32 with PSRAM because pin 16 and 17 don't exist (they are used to access PSRAM), thus we can't use 1-Wire feature with ESP32 model proposing PSRAM.

Note : COM1_RX and COM1_TX can't be used in this test because they can't be short circuited at boot.

### Detailed repro steps so we can see the same problem

1. Plug a temperature sensor ds18b20 following this schematic
[image]
2. Use the 1-Wire sample, you should read 1 device connected with its address
[image]
3. Add the following lines before OneWireController oneWire = new OneWireController();
            Configuration.SetPinFunction(5, DeviceFunction.COM2_RX);
            Configuration.SetPinFunction(18, DeviceFunction.COM2_TX);
4. Keep the sensor on pins 16 and 17, it will find the sensor on those pin despite it's defined on pin 5 and 18
[image]
5. Move the sensor on pins 5 and 18, it will not be found :
[image]

...

### Other suggested things

### Expected behaviour
Using Configuration.SetPinFunction to define other TX2 and RX2 pins should be taken into account by new OneWireController(); and not be hard coded to pins 16 and 17

#### Sample Project
It's the 1-Wire sample with the Configuration.SetPinFunction lines at the beginning :
[1-Wire.zip](https://github.com/nanoframework/Home/files/7994432/1-Wire.zip)

### Screenshot
See above

### Additional context

### I've made an effort to fix the bug

josesimoes, 2022-02-03:
@jefmad 💯 for the detailed bug report! 👏🏻 👏🏻 👏🏻

rkalwak, 2022-02-03:
Additionally I didn't find it documented anywere that OneWire requires pin 16 and pin 17 (RX2 and TX2) only. It is only visible in the ds18b20 sample

jefmad, 2022-02-05:
@josesimoes Could you please instruct where to find the sources for the methods of https://github.com/nanoframework/nanoFramework.Devices.OneWire/blob/develop/nanoFramework.Devices.OneWire/OneWire.cs because they are not accessible and I don't know how to investigate and help further...
I started with building my own OneWire controller using serial port, writing and reading bytes to the sensor, but I think there is an easier solution :-)

Ellerbach, 2022-02-05:
You'll find in each device here for the native part: https://github.com/nanoframework/nf-interpreter/tree/develop/targets
And also the definitions (stub) here to update if you make changes on the managed side: https://github.com/nanoframework/nf-interpreter/tree/develop/src/nanoFramework.Devices.OneWire
You'll need to use the dev containers or local install to rebuild the images and test on devices you have.

Ellerbach, 2022-02-05:
@josesimoes if someone touches this implementation I would then advice to adjust the namespace tot System.Device.OneWire ;-)

jefmad, 2022-02-07:
I managed to make progress in making pins configurable for 1-Wire. Here is the status and then the questions :-)

1- Thanks again for the great doc about dev container, it make things so simple !
2- I managed to build nanoclr with the improvements and tested it on Devkit C, the pin configuration is correctly taken into account and device is found.

Questions : 
1 - Is there a better way to test the modifications made in cpp than 1)building nanoclr, 2)flash it in ESP32 3)See the various result of the printf results 4) adjust the code in cpp and do the process again...?
2 - Now I want to test the result on a ESP32 that has PSRAM (and no pins 16 and 17) In VS Code what is the target I should choose ? What is the one equivalent to EPS32_PSRAM_REV0 ? I can't see in the following picture
3 - The last test I'll perform is using at same time OneWire and PSRAM, to be sure that COM2 is not used for PSRAM, even if I change the pins 16 and 17 to something else. Is there a test or a sample to be sure that ESP32 is using PSRAM ? 

Thanks !
[image]

josesimoes, 2022-02-08:
> 1- Thanks again for the great doc about dev container, it make things so simple ! 2- I managed to build nanoclr with the improvements and tested it on Devkit C, the pin configuration is correctly taken into account and device is found.

That's great! You're on step further into being able to contribute more and have a deeper insight on how all this works.

> Questions : 1 - Is there a better way to test the modifications made in cpp than 1)building nanoclr, 2)flash it in ESP32 3)See the various result of the printf results 4) adjust the code in cpp and do the process again...? 

The alternative is to debug the code directly. For that you need an ESP32 with JTAG connection (such as the WROVER or a DevKit and wiring up an external JTAG.

2 - Now I want to test the result on a ESP32 that has PSRAM (and no pins 16 and 17) In VS Code what is the target I should choose ? 

You can choose any of the several existing ones in the cmake variants template.

What is the one equivalent to ESP32_PSRAM_REV0 ? I can't see in the following picture

Here's the template block; https://github.com/nanoframework/nf-interpreter/blob/70963fd4aafb1539f2b6c0e755abd638d9e6cfb7/targets/ESP32/ESP32/cmake-variants.json#L25

You can either copy&paste it, or run Set-VSCode-CMakes.ps1 in the  install scripts directory.

 3 - The last test I'll perform is using at same time OneWire and PSRAM, to be sure that COM2 is not used for PSRAM, even if I change the pins 16 and 17 to something else. Is there a test or a sample to be sure that ESP32 is using PSRAM ?
PSRAM doesn't use any UART. What happens is that there is an overlap of the default pin assigment, as it has been established.

 Keep up your progress! 😄

jefmad, 2022-02-14:
Thanks @josesimoes  and @Ellerbach it was very useful :-)
Here is my small contribution, it has been tested on DevKit-C and LilyGo. I'm not a developer, so I don't know how to deliver this but sharing the files so that you integrate this modification in the next build.

1. **target_nf_devices_onewire_config.h**  (from /workspaces/nf-interpreter-develop/targets/ESP32/ESP32/) as been modified, you'll see the modification
2. **nf_devices_onewire_native_nanoFramework_Devices_OneWire_OneWireController.cpp** (from /workspaces/nf-interpreter-develop/targets/ESP32/_nanoCLR/nanoFramework.Devices.OneWire/) has few modifications, only in the procedure "**bool oneWireInit()**", a reference to **<Esp32_DeviceMapping.h>** has been added
3. The sample for 1-Wire has been modified with a reference to nanoFramework.Hardware.Esp32 (nuget ref must be included) and the configuration of pins

It all works for me :-) To me you can mark this issue as solved
[Solving issue 961.zip](https://github.com/nanoframework/Home/files/8062224/Solving.issue.961.zip)

josesimoes, 2022-02-14:
@jefmad that's great!! Happy for you. 👏🏻 😄 
If you're not comfortable with raising a PR with your fix, I'll take care of it.

(but I can assure you it won't hurt and it's not _that_ complicated 😋 )

jefmad, 2022-02-17:
Well I'm not sure I'll have some time soon to do it, so I would be nice if you could do it :-) And I'll learn for the next time how to raise a PR ;-)
Thanks !

josesimoes, 2022-02-24:
@jefmad please give it a try with the fw version that's coming out of the pipeline in a few minutes! 😉

closed 2022-02-24 by josesimoes